Unity-Technologies/Unity-Robotics-Hub #50: Missing VHACD repository

siegeledietmar, 2020-11-13:
Hi!

I get the following error opening the Unity project..

> An error occurred while resolving packages:
>   Project has invalid dependencies:
>     com.unity.robotics.urdf-importer: Cannot checkout repository [https://github.com/Unity-Technologies/URDF-Importer.git]:
>       Error when executing git command. Submodule 'VHACD' (https://github.com/Unity-Technologies/VHACD) registered for path 'Runtime/VHACD'
>       Cloning into 'D:/Unity/Unity-Robotics-Hub/tutorials/pick_and_place/PickAndPlaceProject/Library/PackageCache/.tmp7792ZYEX83svpiql/clone/Runtime/VHACD'...
>       remote: Repository not found.
>       fatal: repository 'https://github.com/Unity-Technologies/VHACD/' not found
>       fatal: clone of 'https://github.com/Unity-Technologies/VHACD' into submodule path 'D:/Unity/Unity-Robotics-Hub/tutorials/pick_and_place/PickAndPlaceProject/Library/PackageCache/.tmp7792ZYEX83svpiql/clone/Runtime/VHACD' failed
>       Failed to clone 'Runtime/VHACD'. Retry scheduled
>       Cloning into 'D:/Unity/Unity-Robotics-Hub/tutorials/pick_and_place/PickAndPlaceProject/Library/PackageCache/.tmp7792ZYEX83svpiql/clone/Runtime/VHACD'...
>       remote: Repository not found.
>       fatal: repository 'https://github.com/Unity-Technologies/VHACD/' not found
>       fatal: clone of 'https://github.com/Unity-Technologies/VHACD' into submodule path 'D:/Unity/Unity-Robotics-Hub/tutorials/pick_and_place/PickAndPlaceProject/Library/PackageCache/.tmp7792ZYEX83svpiql/clone/Runtime/VHACD' failed
>       Failed to clone 'Runtime/VHACD' a second time, aborting
> 
> A re-import of the project may be required to fix the issue or a manual modification of D:/Unity/Unity-Robotics-Hub/tutorials/pick_and_place/PickAndPlaceProject/Packages/manifest.json file.

Where can I find the VHACD repro? Thank you!

sarah-gibson, 2020-11-14:
Hi @siegeledietmar,

We are working on a fix to expose the VHACD repo. In the mean time, please use the branch [vidur/hub-update-without-VHACD](https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/vidur/hub-update-without-VHACD) as a workaround. Let us know if you still have issues.

Thanks,
Sarah

siegeledietmar, 2020-11-17:
The branch works fine! Thanks!

vidurvij-Unity, 2020-11-21:
Hi @siegeledietmar,
The dependency has been made public now. You can now switch to the main branch. It uses VHACD for collision mesh decomposition. Let us know if you have more issue.
Vidur

vidurvij-Unity, 2020-12-04:
Hi @siegeledietmar,
We appreciate you bringing this issue to the forefront. We have been able to fix this issue in the latest pull requests.I am closing this issue for now. Feel free open this issue again if this problem problem persists
Vidur

closed 2020-12-04 by vidurvij-Unity