Unity-Technologies/Unity-Robotics-Hub #53: HoloLens and Android compatibility

siegeledietmar, 2020-11-14:
Hi!

I tried to apply the tutorial on an Android smartphone and a HoloLens 2.. both builds show the same strange behavior, that the robot arm is randomly jumping. Any solution for that? HoloLens 2 seems to connect to the ROS node (I get the request in the ROS terminal), but with Android this does not work.

Thank you!

sarah-gibson, 2020-11-16:
Hi @siegeledietmar, 

Are you running the Unity build on Android and HoloLens? Or the ROS side? Or both?

Thanks,
Sarah

siegeledietmar, 2020-11-16:
I am running the Unity build on Android and HoloLens 2. The ROS side is running as in the tutorial. When I run it in the Unity Editor it works perfect, when I compile the same Scene, it shows this strange behaviour. I use the XRSDK.

siegeledietmar, 2020-11-17:
To be more specific, it happens to me for the Pick and Place tutorial.

peifeng-unity, 2020-11-18:
Hi @siegeledietmar,

Thank you for the feedback. We are looking into the issue. In the meantime, may I confirm a couple of things with you:
* Does the "random jumping" issue happen in [Part 1](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/1_urdf.md) of the Pick and Place tutorial or the [Part 3](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/3_pick_and_place.md)?
* You've mentioned an issue that the Unity build cannot communicate with the ROS nodes properly when running on Android. Does it happen in Pick and Place tutorial too?

Thanks,
Peifeng

siegeledietmar, 2020-11-18:
- The issue (for me it seems that the damping and stiffness effects this) can be obtained in Part 1 and 3. It does not appear when running in the Play mode.
- The communication to ROS under Android is not working in the Pick and Place tutorial. In the ROS–Unity Integration tutorial it works fine.

siegeledietmar, 2020-11-18:
The communication problem with Android was my fault, that is working fine now, sorry about that..
The misbehavior of the robot arm exists still one HoloLens 2 and Android.

peifeng-unity, 2020-11-19:
Get you. We are trying to reproduce the issue. Would you mind to share a gif or a short video about what the "random jumping" looks like? It will be very helpful for us to identify the root cause. Thanks!

siegeledietmar, 2020-11-19:
I found in the meantime a simple fix:
Stiffness and Damping values are only set in the _Controller_ script. If I add these values to the X Drive in all _Articulation Body_ it works also as intended on HoloLens and Android.

peifeng-unity, 2021-01-05:
Thank you, @siegeledietmar for the feedback. It is great that you have found a workaround. 

The stiffness and damping values set in the _Controller_ were supposed to be set to each of the articulation bodies in _Start()_ of the Play, and there was a bug in the _URDF-Importer_ before that didn't configure the damping values. The bug has been fixed now in [this PR](https://github.com/Unity-Technologies/URDF-Importer/pull/4) and [the _Controller.cs_](https://github.com/Unity-Technologies/URDF-Importer/blob/main/Runtime/Controller/Controller.cs#L169).

peifeng-unity, 2021-01-05:
I am going to close this issue since the bug is fixed. Please feel free to reopen it if there are more questions or comments. Thank you!

closed 2021-01-05 by peifeng-unity

ANG0-0GNA, 2021-02-04:
> Hi!
>
> I tried to apply the tutorial on an Android smartphone and a HoloLens 2.. both builds show the same strange behavior, that the robot arm is randomly jumping. Any solution for that? HoloLens 2 [...]

Hi,  Dietmar @siegeledietmar , could you please show more details about how to apply this tutorial on hololens 2? I plan to build this tutorial _ROS–Unity Integration: Publisher_ . But after click the build button on unity,  there are many error shown below:
`Assets\RosMessages\RoboticsDemo\msg\PosRot.cs(6,7): error CS0246: The type or namespace name 'RosMessageGeneration' could not be found (are you missing a using directive or an assembly reference?)
Assets\RosMessages\RoboticsDemo\msg\UnityColor.cs(6,7): error CS0246: The type or namespace name 'RosMessageGeneration' could not be found (are you missing a using directive or an assembly reference?)
Assets\RosMessages\RoboticsDemo\msg\UnityColor.cs(10,31): error CS0246: The type or namespace name 'Message' could not be found (are you missing a using directive or an assembly reference?)
Assets\Scripts\RosPublisherExample.cs(10,5): error CS0246: The type or namespace name 'ROSConnection' could not be found (are you missing a using directive or an assembly reference?)
Assets\RosMessages\RoboticsDemo\msg\PosRot.cs(43,38): error CS0115: 'PosRot.SerializationStatements()': no suitable method found to override
....
....
....
`
In Ros Sharp UWP version, I find these are integrated in some plugins, such as RosBridgeClientUWP.dll, etc. I don't know how it works in this project on hololens 2. So could you show the details about how to apply this tutorial on hololens 2? Thank you very much!