Unity-Technologies/Unity-Robotics-Hub #61: Articulation XDrive cannot rotate

PolarisYxh, 2020-11-25:
[image]
I want to rotate my robote's last three fingers, but give values in range to three fingers' XDrive's target cannot move fingers. Other parents' body can be moved(stiffnes and other param is same), I think may be these fingers don't have child articulation body so I add child body to three finges then it works! it's may be a bug. Hope you can resolve this!

LaurieCheers-unity, 2020-11-25:
Hi, thanks for the report. Can you send us your URDF file so we can investigate this bug?

PolarisYxh, 2020-11-26:
`<?xml version="1.0" ?>

<robot name="j2n6s300" xmlns:body="http://playerstage.sourceforge.net/gazebo/xmlschema/#body" xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller" xmlns:gazebo="http://playerstage.sourceforge.net/gazebo/xmlschema/#gz" xmlns:geom="http://playerstage.sourceforge.net/gazebo/xmlschema/#geom" xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface" xmlns:joint="http://playerstage.sourceforge.net/gazebo/xmlschema/#joint" xmlns:model="http://playerstage.sourceforge.net/gazebo/xmlschema/#model" xmlns:physics="http://playerstage.sourceforge.net/gazebo/xmlschema/#physics" xmlns:renderable="http://playerstage.sourceforge.net/gazebo/xmlschema/#renderable" xmlns:rendering="http://playerstage.sourceforge.net/gazebo/xmlschema/#rendering" xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor" xmlns:xi="http://www.w3.org/2001/XInclude">
  
  
  
  
  <link name="root"/>
  
  <link name="world"/>
  <joint name="connect_root_and_world" type="fixed">
    <child link="root"/>
    <parent link="world"/>
    <origin rpy="0 0 0" xyz="0 0 0.5"/>
  </joint>
  
  <link name="j2n6s300_link_base">
    <visual>
      <geometry>
        <mesh filename="package://kinova_description/meshes/base.dae"/>
      </geometry>
      <material name="carbon_fiber">
        <color rgba="0.792156862745098 0.819607843137255 0.933333333333333 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://kinova_description/meshes/base.dae"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.46784"/>
      <origin rpy="0 0 0" xyz="0 0 0.1255"/>
      <inertia ixx="0.000951270861568" ixy="0" ixz="0" iyy="0.000951270861568" iyz="0" izz="0.00037427200000000004"/>
    </inertial>
  </link>
  <joint name="j2n6s300_joint_base" type="fixed">
    <parent link="root"/>
    <child link="j2n6s300_link_base"/>
    <axis xyz="0 0 0"/>
    <limit effort="0" lower="0" upper="0" velocity="0"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <dynamics damping="0.0" friction="0.0"/>
  </joint>
  <link name="j2n6s300_link_1">
    <visual>
      <geometry>
        <mesh filename="package://kinova_description/meshes/shoulder.dae"/>
      </geometry>
      <material name="carbon_fiber">
        <color rgba="0.792156862745098 0.819607843137255 0.933333333333333 1"/>
      </material>
    </visual>
    <visual>
      <geometry>
        <mesh filename="package://kinova_description/meshes/ring_big.dae"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://kinova_description/meshes/shoulder.dae"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.7477"/>
      <origin xyz="0 -0.002 -0.0605"/>
      <inertia ixx="0.0015203172520400004" ixy="0" ixz="0" iyy="0.0015203172520400004" iyz="0" izz="0.00059816"/>
    </inertial>
  </link>
  <joint name="j2n6s300_joint_1" type="continuous">
[... 511 lines cut ...]
  <joint name="j2n6s300_joint_finger_tip_3" type="revolute">
    <parent link="j2n6s300_link_finger_3"/>
    <child link="j2n6s300_link_finger_tip_3"/>
    <axis xyz="0 0 1"/>
    <origin rpy="0 0 0" xyz="0.044 -0.003 0"/>
    <limit effort="2" lower="0" upper="2" velocity="1"/>
  </joint>
  <transmission name="j2n6s300_joint_finger_tip_3_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="j2n6s300_joint_finger_tip_3">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </joint>
    <actuator name="j2n6s300_joint_finger_tip_3_actuator">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>
  <gazebo>
    <plugin filename="libgazebo_ros_control.so" name="gazebo_ros_control">
      <robotNamespace>j2n6s300</robotNamespace>
      <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
      <legacyModeNS>true</legacyModeNS>
    </plugin>
  </gazebo>
</robot>`

PolarisYxh, 2020-11-26:
Files is in
https://drive.google.com/drive/folders/15ayaObwqFfQZWbx3hzOl4gWpxNnT4g4z?usp=sharing
Thank you.

vidurvij-Unity, 2021-01-20:
@PolarisYxh  This is an issue related to the moment of inertia in the Articulation Body. In case of low values of moment of inertia in the articulation body, not enough force is generated to move the joint. You can experiment by increasing the value of moment of inertia in that articulation body.
When you add another articulation body at the end of that joint, you are effectively increasing the moment of inertia on the original joint which in turn increase the force generated by the joint.

vidurvij-Unity, 2021-02-25:
Hi @PolarisYxh,
Were you able to solve the issue ?
Vidur

closed 2021-03-12 by vidurvij-Unity