Unity-Technologies/Unity-Robotics-Hub #63: Target not reachable by the robot

SDBonhof, 2020-11-27:
Hi,

I am following the pick-and-place tutorial. I have followed every step and have encountered no problems up until the very last step. When clicking the 'publish' button in Unity to get the robot to perform the pick and place action, I get the following output in my terminal:

```
You can start planning now!

ROS-Unity Handshake received, will connect to 192.168.2.7:5005
[ INFO] [1606469570.519531449]: Loading robot model 'niryo_one'...
[ INFO] [1606469570.519943415]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1606469570.541337238]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1606469571.595206419]: Ready to take commands for planning group arm.
[ INFO] [1606469571.825751695]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ WARN] [1606469571.825966961]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[ INFO] [1606469571.826019492]: Planning attempt 1 of at most 1
[ INFO] [1606469571.829529746]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1606469571.830429183]: RRTConnect: Starting planning with 1 states already in datastructure
[ERROR] [1606469576.839300394]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1606469576.839339522]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1606469576.839352531]: No solution found after 5.009120 seconds
[ INFO] [1606469576.840151143]: Unable to solve the planning problem
[ INFO] [1606469576.840341969]: ABORTED: No motion plan found. No execution attempted.
[ERROR] [1606469576.840967]: Error processing request: 
            Trajectory could not be planned for a destination of position: 
  x: 0.153000488877
  y: 0.264993906021
  z: 0.7437190786
orientation: 
  x: 0.5
  y: 0.5
  z: -0.5
  w: 0.5 with starting joint angles position: 
  x: 0.153000488877
  y: 0.264993906021
  z: 0.7437190786
orientation: 
  x: 0.5
  y: 0.5
  z: -0.5
  w: 0.5.
            Please make sure target and destination are reachable by the robot.
        
['Traceback (most recent call last):\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 632, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/stefan/unity_ws/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/mover.py", line 89, in plan_pick_and_place\n    pre_grasp_pose = plan_trajectory(move_group, req.pick_pose, current_robot_joint_configuration)\n', '  File "/home/stefan/unity_ws/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/mover.py", line 53, in plan_trajectory\n    raise Exception(exception_str)\n', 'Exception: \n            Trajectory could not be planned for a destination of position: \n  x: 0.153000488877\n  y: 0.264993906021\n  z: 0.7437190786\norientation: \n  x: 0.5\n  y: 0.5\n  z: -0.5\n  w: 0.5 with starting joint angles position: \n  x: 0.153000488877\n  y: 0.264993906021\n  z: 0.7437190786\norientation: \n  x: 0.5\n  y: 0.5\n  z: -0.5\n  w: 0.5.\n            Please make sure target and destination are reachable by the robot.\n        \n']
Service Exception raised. Attempt: 1
```

My environment in Unity looks like this:
[image]
It looks like the robot should be easily able to reach the target. I have tried to move the objects but the issue persists.

The positions of the entities in Unity are:
```
target: -0.265, 0.644, 0.153
targetplacement: 0.281, 0.640, 0.124
nitryo_one (base_link): 0, 0.63, 0
Table: 0, 0, 0
```

It seems like the target is converted correctly from Unity to ROS (X,Y,Z) to (Z, -X, Y)

I am working on Ubuntu 18.04 and Unity 2020.2.0b9.

Any clue why this is happening? Thanks!

SDBonhof, 2020-11-27:
I managed to fix it myself. I raised the target a bit so that it falls onto the table when I press play. Perhaps the target was clipped into the table slightly or something. Still not sure why that would be a problem for goal reachability but at least it works now.

LaurieCheers-unity, 2020-11-27:
Thanks for the feedback; glad you were able to fix the issue. The Robotics Hub project is under active development and we will be looking into this case to see why it wasn't working initially.

NathanielRose, 2020-11-29:
@SDBonhof What did you raise the target to. Its seems as if I am having a similar experience.

SDBonhof, 2020-11-30:
@NathanielRose It is a bit sensitive, but I can get the tutorial working when I place the target at
```
-0.218, 0.805, 0.199
```

Upon further inspection, placing the target at the same x,z coordinates but lowered onto (or even slightly into) the table still results in a successful pick and place. Restarting the ROS nodes with the launch file did not change the outcome. 

Placing the target back at my original locations as shown in my original post results in a failing pick and place again.

It seems to me like the robot is unable to reach targets that are on its left. Placing a target at around 8 o'clock causes the robot to swing left to go to the target and reaches it. But placing it closer to 9 o'clock causes the pick to fail. from about 9.30-10-ish towards 12 the robot is able to reach the target again. 

I find this quite strange behaviour since the robot is able to swing past 9 to 8 (from 12) but it supposedly isn't able to reach 9?

mrpropellers, 2020-11-30:
@SDBonhof Collisions with the table should be a non-issue, as the MoveIt solver, which is computing the trajectories, doesn't know of the table's existence, so whatever is happening here is not necessarily the robot failing to reach the location, but the solver failing to compute a valid path for those start/end points.  

I'm going to log your observation that the solver fails specifically when the Target is on the left and TargetPlacement is on the right - that seems to be the key factor is what causes the path-planning to fail... Could you confirm, when setting up the scene initially, did you drag the Target and TargetPlacement objects into the Scene view, placing them on the table manually, or into the Hierarchy?  When dragged into the Hierarchy, they should show up in a default location in the view that is (presumably) guaranteed to work, and should look something like this:
[image]

SDBonhof, 2020-12-01:
@mrpropellers What you say about the MoveIt planner makes sense. 

IIRC I dragged the objects into the scene directly instead of into the hierarchy. Nevertheless, the location should be reachable either way, so I am happy you are looking into it.

mrpropellers, 2021-02-01:
Just to provide an update: this issue has been noted internally but it hasn't made it to the top of the list yet. There's a few tasks ahead of it but I will find time soon to try and root cause this and determine whether it's an issue within our stack that we can fix. If it's a problem inherent to move-it we can at least note the exact conditions under which planning would be expected to fail.

SDBonhof, 2021-02-02:
Thanks for the update. For me this issue isn't pressing as I am not using Unity at the moment, but maybe others experience this issue too, idk.

mrpropellers, 2021-02-17:
So I finally had some time to check in on this and was unable to reproduce.  I was able to identify a region directly behind the arm that was (correctly) reported as unreachable, but with Ubuntu 18.04 and the Target and Placement objects at the coordinates specified, trajectory planning succeeded:
[image]

While investigating, I saw there was a confounding issue which caused the trajectory planner to break when it failed to find a path so I've submitted a fix for that: #155  

Hopefully this issue was resolved incidentally by some previous code change. I'm going to close since we can't repro, but if this crops up again feel free to either re-open or submit a new Issue!

closed 2021-02-17 by mrpropellers