Unity-Technologies/Unity-Robotics-Hub #66: TCP Socket Exception

LikithRana, 2020-12-02:
Hello,

I am following the pick-and-place tutorial which comprises of 3 parts. I have followed every step and have encountered no problems up until the very last step of the Part 2. When I click on the 'publish' button in the game mode of Unity, I am getting the following exception:
[image]

On the ROS side, "roslaunch" command is running but it doesn't print any data :
[image]

I have disabled firewall and tried too but no success. Can you please let me know if I am missing something? I apologize if this is a simple issue but I am new to this. Hence I am not sure what else can I do next to figure it out? Thank you.

mrpropellers, 2020-12-02:
Hi Likith! This looks to be an issue with how your network adapters are set up in your virtual machine. You may need to do some troubleshooting to ensure that your guest OS can talk to your host OS and vice versa. One simple way to do this is to set up a "Host-Only" style network, but this varies based on what virtualization software you're using and you'll have to do a little searching around the internet to find a good solution for your setup.  Try and ensure that you can `ping` your host OS's IP from inside the guest, and can `ping` the guest from the host, then try this last step again. If the ping works, but the ROS communication still fails, then there may be an issue to investigate here, but my hunch is that there is just some additional configuration needed for your VM's network settings.

LikithRana, 2020-12-03:
Hello @mrpropellers, It worked once I setup "Host-Only" network in my Virtual Machine. Thank you very much for the quick response and your help.

Just for reference, I followed the below links to setup "Host-only" network:
https://serverfault.com/questions/225155/virtualbox-how-to-set-up-networking-so-both-host-and-guest-can-access-internet
https://www.youtube.com/watch?v=73ccYLqGb_k

mrpropellers, 2020-12-03:
Great!  That will be a useful resource for us to pass to future users that encounter similar issues.  Since it seems like your issue is resolved, I'm going to go ahead and close this ticket, thanks!

closed 2020-12-03 by mrpropellers

mrpropellers, 2021-03-10:
Don't mind me, just adding some additional text to make this issue more searchable:

> SocketException: A connection attempt failed because the connected party did not respond after a period of time, or established connection failed because connected host has failed to respond
TCPConnector Exception: System.Net.Sockets.SocketException: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.