Unity-Technologies/Unity-Robotics-Hub #67: ReadME Update - Pick-and-Place Tutorial: Part 2 - The ROS side

nullbyte91, 2020-12-04:
Hi,

There are two ros msg package[ros-melodic-octomap-msgs, ros-melodic-object-recognition-msgs] need to install to compile src code by using catkin_make. 

We can update this in README.

-
Jegathesan S

mrpropellers, 2020-12-04:
Hi Jegathesan, can you elaborate a bit on why you found these messages were needed? Do you know specifically which catkin package was failing to compile without them included?

nullbyte91, 2020-12-05:
@mrpropellers I'm using Ubuntu 18.04 and We need to install [ros-melodic-octomap-msgs, ros-melodic-object-recognition-msgs] these two packages to compile the ROS source code by using catkin_make.

We need to install [ ros-melodic-moveit, ros-melodic-joint-trajectory-controller] to run the part 3 launch file[roslaunch niryo_moveit part_3.launch].

mrpropellers, 2020-12-07:
@nullbyte91 When you say "compile the ROS source code" are you talking about compiling your initial ROS environment from source?  What does "the ROS source code" refer to in this context? All the dependencies you're listing seem to be standard ros packages that should be installed by apt-get on initial setup. Can you confirm what your original install target was when you installed ROS?

nullbyte91, 2020-12-09:
@mrpropellers Yes, I was referring the initial ROS environment from source. And, Yes these packages can be installed by using apt package manager. Since these packages not part of ros-melodic-desktop-full or base we can mentioned it in README.

mrpropellers, 2020-12-09:
@nullbyte91 Instructions for installing ros-melodic-moveit and ros-melodic-join-trajectory-controller already exist in the ROS setup instructions of Part 2: https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/2_ros_tcp.md#manually-setup-ros

Note the line:
```bash
sudo apt-get install python-pip ros-melodic-robot-state-publisher ros-melodic-moveit ros-melodic-rosbridge-suite ros-melodic-joy ros-melodic-ros-control ros-melodic-ros-controllers ros-melodic-tf2-web-republisher```
where `ros-melodic-moveit` is installed explicitly, and `ros-melodic-joint-trajectory-controller` is installed as a dependency of `ros-melodic-ros-controllers`.  `ros-melodic-octomap-msgs` and `ros-melodic-object-recognition-msgs` are also installed as dependencies of `ros-melodic-moveit-msgs`.  What am I missing?

closed 2020-12-10 by nullbyte91