Unity-Technologies/Unity-Robotics-Hub #69: 3d Lidar to pointcloud2 and publish

ironWolf1990, 2020-12-04:
I am not sure if this is the right place to put this question. I am working on a 3D lidar sensor implementation ... 

```
private void FixedUpdate()
{
    for (int i = 0; i < samples; i++)
    {
        for (int y = 0; y < numberOfLayers; y++)
        {
            float angle = minVerticalAngle + (float)y * vertIncrement;
            directions = transform.rotation * Quaternion.Euler(-angle, azimutIncrAngle * i, 0) * transform.forward;

            raycastHits = new RaycastHit();
            if (Physics.Raycast(transform.position, directions, out raycastHits, rangeMax))
            {
                if (raycastHits.distance >= rangeMin && raycastHits.distance <= rangeMax)
                {
                    ranges = raycastHits.distance;
                    Debug.DrawRay(transform.position, directions * ranges, Color.red);
                }
            }
            else
            {
                Debug.DrawRay(transform.position, directions * rangeMax, Color.green);
            }
        }
    }
}
```

After saving the points, I want to publish it to a ros node as a pointcloud2 message. I looked at [Velodyne Implementation by vwaurich](https://github.com/vwaurich/VelodyneLidarUnitySimulation/blob/master/Scripts/Velodyne_UDP_Send.cs) and [
GazeboRosVelodyneLaser.cpp](https://bitbucket.org/DataspeedInc/velodyne_simulator/src/master/velodyne_gazebo_plugins/src/GazeboRosVelodyneLaser.cpp) but couldn't wrap my head around it! Can someone kindly guide me on this issue ?
Thank you in advance.

mrpropellers, 2020-12-04:
Hi IronWolf - can you tell me me more about which part of your implementation you are having issues with? Also, is it a requirement that your point cloud be published as a PointCloud2 message, as opposed to PointCloud?  PointCloud2 tends to require a deeper understanding of how you're packing your data, and would also require you to convert more ROS message types into C# objects than PointCloud would.

ironWolf1990, 2020-12-05:
Hey @mrpropellers , honestly i am new to ros and robotics. I picked up unity few months ago and was having fun with procedural generation. After reading the [blog](https://blogs.unity3d.com/2020/11/19/robotics-simulation-in-unity-is-as-easy-as-1-2-3/) i got very much interested in robotics and am working on a personal hobby project.<br/>
After going over ros's documentation i [read](https://wiki.ros.org/pcl/Overview) `The newly revised ROS point cloud message (and currently the de facto standard in PCL), ...`. I've adapted most off the messaging `publishing` and `subscribing`  but got stuck with 3d map data , in this case 3D Lidar.  

Was hoping to get some pointers on how to tackle this problem.

ironWolf1990, 2020-12-06:
I ended up using pointcloud instead of pointcloud2 at the moment.

closed 2020-12-06 by ironWolf1990

mrpropellers, 2020-12-07:
Cool!  It may be the case that some PCL tools require PointCloud2 messages, but if you don't have a driving need for your experiment to be performative, it'd likely be easier to write a simple transformer on the ROS side of things to subscribe to your PointCloud output from Unity and convert it to PointCloud2. If you're working with the C++ PCL libraries, it's likely a pretty straightforward memcpy to repack the data. Good luck and feel free to reopen if there is something else on the Unity side of things that we need to address!