Unity-Technologies/Unity-Robotics-Hub #73: Index out of array bounds when accessing prismatic joint position

HenningLuo, 2020-12-12:
Hi,

I was trying to publish my robot states using ros connector. It worked properly with revolute joints, however, when I tried to access the states of primatic finger joints, I got this error:
`
IndexOutOfRangeException: Index was outside the bounds of the array.
UnityEngine.ArticulationReducedSpace.get_Item (System.Int32 i) (at /home/bokken/buildslave/unity/build/Modules/Physics/ScriptBindings/Dynamics.bindings.cs:964)
RosSharp.Urdf.UrdfJointPrismatic.GetPosition () (at Library/PackageCache/com.unity.robotics.urdf-importer@ca9f90f7d3/Runtime/UrdfComponents/UrdfJoints/UrdfJointPrismatic.cs:51)
JointStatePublisher.FixedUpdate () (at Assets/Scripts/JointStatePublisher.cs:102)
`
Here is how I accessed the states:
`
message.position[i+numJoints] = UrdfPrismaticJoints[i].GetPosition();
message.velocity[i+numJoints] = UrdfPrismaticJoints[i].GetVelocity();
message.effort[i+numJoints] = UrdfPrismaticJoints[i].GetEffort();
`
After checking the code on urdf prismatic joints in UrdfJointPrismatic.cs, I found the source of this error:
`
        public override float GetPosition()
        {
            #if UNITY_2020_1_OR_NEWER
                return unityJoint.jointPosition[3];
            #else
            return Vector3.Dot(unityJoint.transform.localPosition - unityJoint.connectedAnchor, unityJoint.axis);
            #endif
        }
`
Apperantly the GetPosition is returning the 4th dimension of the joint position, which led to the aforementioned error. Could this by any chance be a bug? Or if it is not, could anybody kindly share some ideas on how this might be resolved? Any thoughts are appreciated. Thanks in advance!

at669, 2020-12-14:
Hi @HenningLuo,

Thanks for the feedback! We will investigate this further. In the interim, try defaulting to the `jointPosition[0]` on the desired Articulation Body of the prismatic joint, i.e.:

```csharp
var prismaticJointBody = desired.GetComponent<ArticulationBody>();
message.position[i+numJoints] = prismaticJointBody.jointPosition[0]; 
```

Let us know if that resolves your issue and gives the desired value.

HenningLuo, 2020-12-14:
Hi @at669,

Thanks! It works well!

closed 2020-12-14 by HenningLuo