Unity-Technologies/Unity-Robotics-Hub #74: Extending the demo

ngunsu, 2020-12-14:
I am interested in publishing another type of sensor in ROS, however, I am not sure of how to it. For example, I want to add a sensor_msgs/CompressedImage, how can I do it?. I understand the basics of message generation, but I don't get how to convert the camera to the requesting input type.

HenningLuo, 2020-12-14:
Hi, 

you can create your image type in this way
```
private RosMessageTypes.Sensor.CompressedImage message;
```
Fill you data into the message in every update following the data structure of [CompressedImage](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CompressedImage.html). It should be straight forward to you.

Best

at669, 2020-12-14:
Hi @ngunsu,

If you want to grab an image from a Unity Camera to send as a sensor_msgs/CompressedImage, you will need to convert the Camera data into a `uint8[] data` format for the CompressedImage data structure @HenningLuo mentioned. One method to do this is to create and set a new [Render Texture](https://docs.unity3d.com/ScriptReference/RenderTexture.html) as your desired sensor camera's target texture, `ReadPixels` and `GetRawTextureData` from the this texture, and then convert it to the appropriate `byte[]` format. Some template code is below for reference:

```csharp
// Grab the texture data from the desired camera: Camera.main in this example
var sensorCamera = Camera.main;
var currentRT = RenderTexture.active;
RenderTexture.active = sensorCamera.targetTexture;
sensorCamera.Render();
Texture2D camText = new Texture2D(sensorCamera.targetTexture.width, sensorCamera.targetTexture.height);
camText.ReadPixels(new Rect(0, 0, sensorCamera.targetTexture.width, sensorCamera.targetTexture.height), 0, 0);
camText.Apply();
RenderTexture.active = currentRT;

// Convert the texture data to bytes
byte[] imageBytes = camText.GetRawTextureData();
var message = new RosMessageTypes.Sensor.CompressedImage(new RosMessageTypes.Std.Header(), "png", imageBytes);
```

Let us know if this solves your issue!

ngunsu, 2020-12-14:
This solves my question, thanks

at669, 2020-12-14:
Great! Since the issue is resolved, I will close it out. Thanks!

closed 2020-12-14 by at669