Unity-Technologies/Unity-Robotics-Hub #75: problem: docker run to run ROS in a new docker container

eeGeoGZP, 2020-12-16:
Hi,

I got Unity-Robotics-Hub updated this morning (got docker folder in \Unity-Robotics-Hub\tutorials\pick_and_place) on my windows10 laptop. then I follow the instructions in tutorial Part 2 -> The ROS Side -> Use Docker Container. I can finish step “2. Build the ROS docker image” without problem. But there is error when I do 3. 

Here are the steps I did:
1. cd C:\Users\Public\Documents\UnityProjects2020\RobotSim\Unity-Robotics-Hub\tutorials\pick_and_place
2. docker build -t unity-robotics:pick-and-place -f docker/Dockerfile .
3. docker run -it --rm -p 10000:10000 -p 5005:5005 unity-robotics:pick-and-place part_2 /bin/bash
then it keeps restarting like this [server_endpoint-2] restarting process 
and the error is
_/usr/bin/env: ‘python\r’: No such file or directory

The output in the shell window is
C:\Users\Public\Documents\UnityProjects2020\RobotSim\Unity-Robotics-Hub\tutorials\pick_and_place>docker run -it --rm -p 10000:10000 -p 5005:5005 unity-robotics:pick-and-place part_2 /bin/bash
... logging to /root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/roslaunch-2813e20d274d-32.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://2813e20d274d:38871/

SUMMARY
========

PARAMETERS
 * /ROS_IP: 172.17.0.2
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    server_endpoint (niryo_moveit/server_endpoint.py)
    trajectory_subscriber (niryo_moveit/trajectory_subscriber.py)

auto-starting new master
process[master]: started with pid [42]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6baeae64-3fbf-11eb-b080-0242ac110002
process[rosout-1]: started with pid [53]
started core service [/rosout]
process[server_endpoint-2]: started with pid [56]
/usr/bin/env: ‘python\r’: No such file or directory
[server_endpoint-2] process has died [pid 56, exit code 127, cmd /catkin_ws/src/niryo_moveit/scripts/server_endpoint.py --wait __name:=server_endpoint __log:=/root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2.log].
log file: /root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2*.log
[server_endpoint-2] restarting process
/usr/bin/env: ‘python\r’: No such file or directory
process[trajectory_subscriber-3]: started with pid [61]
process[server_endpoint-2]: started with pid [62]
/usr/bin/env: ‘python\r’: No such file or directory
[server_endpoint-2] process has died [pid 62, exit code 127, cmd /catkin_ws/src/niryo_moveit/scripts/server_endpoint.py --wait __name:=server_endpoint __log:=/root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2.log].
log file: /root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2*.log
[trajectory_subscriber-3] process has died [pid 61, exit code 127, cmd /catkin_ws/src/niryo_moveit/scripts/trajectory_subscriber.py --wait __name:=trajectory_subscriber __log:=/root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/trajectory_subscriber-3.log].
log file: /root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/trajectory_subscriber-3*.log
[server_endpoint-2] restarting process
process[server_endpoint-2]: started with pid [63]
/usr/bin/env: ‘python\r’: No such file or directory
[server_endpoint-2] process has died [pid 63, exit code 127, cmd /catkin_ws/src/niryo_moveit/scripts/server_endpoint.py --wait __name:=server_endpoint __log:=/root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2.log].
log file: /root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2*.log
[server_endpoint-2] restarting process
process[server_endpoint-2]: started with pid [64]
/usr/bin/env: ‘python\r’: No such file or directory
[server_endpoint-2] process has died [pid 64, exit code 127, cmd /catkin_ws/src/niryo_moveit/scripts/server_endpoint.py --wait __name:=server_endpoint __log:=/root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2.log].
log file: /root/.ros/log/6baeae64-3fbf-11eb-b080-0242ac110002/server_endpoint-2*.log

peifeng-unity, 2020-12-16:
Hello @eeGeoGZP,

Thank you for bringing this issue up. The error seems to show that the "server_endpoint.py" contains DOS carriage return "\r", which cannot be executed in the Ubuntu system inside the docker.

To address the issue, we will need to add a line in Dockerfile to convert line breaks in all the executable files, like https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/add-dos2unix-in-dockerfile/tutorials/pick_and_place/docker/Dockerfile#L19.

We will go and merge the fix. If you need to have it work in the meantime, you can just follow the above Dockerfile and rebuild the docker image.

Please let us know if your problem could be resolved with the fix?

Thank you,
Peifeng

eeGeoGZP, 2020-12-17:
Hi Peifeng, I have added the line RUN /bin/bash -c "find $ROS_WORKSPACE -type f -print0 | xargs -0 dos2unix" at the line #19 as you mentioned. After rebuilt the docker image, "docker run ------" is running ok (without error).

peifeng-unity, 2020-12-17:
Thank you, @eeGeoGZP. I'll close this issue.

closed 2020-12-17 by peifeng-unity