Unity-Technologies/Unity-Robotics-Hub #80: Wrong mesh files in niryo_one.urdf

JPABotermans, 2020-12-21:
Hi,

When looking at the niryo_one.urdf file, it seems that for the grippers the wrong clamps are attached. Is this correct?

[image]

vidurvij-Unity, 2020-12-23:
Hi @JPABotermans ,
Thank you for bringing this to our attention. The mesh files are identical and functionally are working correctly. We will update the naming scheme in a future pull request.
Vidur

peifeng-unity, 2021-01-04:
Hi @JPABotermans, [the fix](https://github.com/Unity-Technologies/Unity-Robotics-Hub/commit/3e229040e55d915d0b377156243d3933b770ab16) has been merged to the `dev` branch, and it will soon be merged into `main`.

I am going to close this issue. Please feel free to reopen it if there are more questions. Thank you!

-Peifeng

closed 2021-01-04 by peifeng-unity