Unity-Technologies/Unity-Robotics-Hub #83: Exception raised!! System.OverflowException  at ......

slxxqs, 2020-12-28:
Hi,  When I Press the UI Button to send the joint configurations to ROS in Pick-and-Place Tutorial: Part 3,there is an error in unity console:
[image]

hyounesy, 2020-12-28:
Hi @slxxqs, Thank you for bringing this to our attention. We will investigate the issue.

mpinol, 2020-12-29:
Hi slxxqs,

I have a few questions,

Does this happen every time you attempt to send the joint configurations?
Are there any errors or relevant log messages on any of the ROS windows?
Does the returned trajectory contain any data or does it only contain default empty values?

In the meantime I am going to try to reproduce the error and will let you know if I find anything.

slxxqs, 2020-12-31:
Hi,@mpinol 
1. It happened every time I attempt to send the joint configurations.
2. When I first run the following roslaunch `roslaunch niryo_moveit part_3.launch`,there is an error: `unused args [urdf_without_meshes] for include of [Unity-Robotics-Hub-main/tutorials/pick_and_place/ROS/src/niryo_moveit/launch/niryo_one_base.launch] The traceback for the exception was written to the log file`.
so I deleted `    <arg name="urdf_without_meshes" default="false" />` in demo.launch.
When I send the joint configurations, MoveIt received the planning request and computed several motion plans. There is no error message on ROS windows.
3. After the motion planning, there is an error in unity 

>Exception raised!! System.OverflowException
  at (wrapper managed-to-native) System.Object.__icall_wrapper_ves_icall_array_new_specific(intptr,int)
  at RosMessageTypes.NiryoMoveit.MoverServiceResponse.Deserialize (System.Byte[] data, System.Int32 offset) [0x0000e] in G:\Unity-Robotics-Hub-main\tutorials\pick_and_place\PickAndPlaceProject\Assets\RosMessages\NiryoMoveit\srv\MoverServiceResponse.cs:40 
  at ROSConnection+<SendServiceMessage>d__19'1[RESPONSE].MoveNext () [0x002b3] in G:\Unity-Robotics-Hub-main\tutorials\pick_and_place\PickAndPlaceProject\Library\PackageCache\com.unity.robotics.ros-tcp-connector@8084143f76\Runtime\TcpConnector\ROSConnection.cs:133 
UnityEngine.Debug:LogError (object)
ROSConnection/<SendServiceMessage>d__19'1<RosMessageTypes.NiryoMoveit.MoverServiceResponse>:MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@8084143f76/Runtime/TcpConnector/ROSConnection.cs:137)
UnityEngine.UnitySynchronizationContext:ExecuteTasks ()
[image]

mrpropellers, 2021-01-20:
Hi @slxxqs in the interest keeping our backlog up to date can I ask if you're still encountering this issue? We're happy to investigate if you can provide some more information about your environment, or if this is no longer an issue, could you provide a quick summary of what changed so we can resolve this ticket?

closed 2021-01-21 by slxxqs

zhaozhongch, 2021-02-13:
I met that error too 
```
unused args [urdf_without_meshes] for include of [Unity-Robotics-Hub-main/tutorials/pick_and_place/ROS/src/niryo_moveit/launch/niryo_one_base.launch] 
```
Seems the if the arg is not used we should not pass it to `niryo_one_base.launch`. Comment or delete the line works.