Unity-Technologies/Unity-Robotics-Hub #85: Docker Image Failing during build in ROS Tutorial

NathanielRose, 2021-01-01:
Hello,

It looks like the latest DockerFile provided in the [ROS Setup](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/2_ros_tcp.md#the-ros-side) is not building correctly on my machine.

I am receiving the current error in the 12th step when I run `docker build -t unity-robotics:pick-and-place -f docker/Dockerfile .`

Am I missing something?

```
 => CACHED [11/12] RUN /bin/bash -c "find /catkin_ws -type f -prin  0.0s
 => ERROR [12/12] RUN dos2unix /tutorial && dos2unix /setup.sh &&   8.4s
------
 > [12/12] RUN dos2unix /tutorial && dos2unix /setup.sh && chmod +x /setup.sh && /setup.sh && rm /setup.sh:
#17 0.700 dos2unix: converting file /tutorial to Unix format...
#17 0.702 dos2unix: converting file /setup.sh to Unix format...
#17 1.119 -- The C compiler identification is GNU 7.5.0
#17 1.221 -- The CXX compiler identification is GNU 7.5.0
.
.
.
#17 7.335 -- Generating .msg files for action moveit_msgs/Pickup /catkin_ws/src/moveit_msgs/action/Pickup.action
#17 7.354 Generating for action Pickup
#17 7.357 -- Generating .msg files for action moveit_msgs/Place /catkin_ws/src/moveit_msgs/action/Place.action
#17 7.384 Generating for action Place
#17 7.938 -- moveit_msgs: 81 messages, 24 services
#17 8.271 -- +++ processing catkin package: 'niryo_moveit'
#17 8.271 -- ==> add_subdirectory(niryo_moveit)
#17 8.301 -- Could NOT find ros_tcp_endpoint (missing: ros_tcp_endpoint_DIR)
#17 8.301 -- Could not find the required component 'ros_tcp_endpoint'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
#17 8.301 CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
#17 8.301   Could not find a package configuration file provided by "ros_tcp_endpoint"
#17 8.301   with any of the following names:
#17 8.301 
#17 8.301     ros_tcp_endpointConfig.cmake
#17 8.301     ros_tcp_endpoint-config.cmake
#17 8.301 
#17 8.301   Add the installation prefix of "ros_tcp_endpoint" to CMAKE_PREFIX_PATH or
#17 8.301   set "ros_tcp_endpoint_DIR" to a directory containing one of the above
#17 8.301   files.  If "ros_tcp_endpoint" provides a separate development package or
#17 8.301   SDK, be sure it has been installed.
#17 8.301 Call Stack (most recent call first):
#17 8.301   niryo_moveit/CMakeLists.txt:4 (find_package)
#17 8.301 
#17 8.301 
#17 8.307 -- Configuring incomplete, errors occurred!
#17 8.307 See also "/catkin_ws/build/CMakeFiles/CMakeOutput.log".
#17 8.307 See also "/catkin_ws/build/CMakeFiles/CMakeError.log".
#17 8.314 Invoking "cmake" failed
#17 8.314 Base path: /catkin_ws
#17 8.314 Source space: /catkin_ws/src
#17 8.314 Build space: /catkin_ws/build
#17 8.314 Devel space: /catkin_ws/devel
#17 8.314 Install space: /catkin_ws/install
#17 8.314 Creating symlink "/catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
#17 8.314 ####
#17 8.314 #### Running command: "cmake /catkin_ws/src -DCATKIN_DEVEL_PREFIX=/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/catkin_ws/install -G Unix Makefiles" in "/catkin_ws/build"
#17 8.314 ####
------
executor failed running [/bin/sh -c dos2unix /tutorial && dos2unix /setup.sh && chmod +x /setup.sh && /setup.sh && rm /setup.sh]: exit code: 1
```

peifeng-unity, 2021-01-04:
Hello @NathanielRose, thank you for trying our ROS docker tutorial. From the log, the issue seems `Could not find a package configuration file provided by "ros_tcp_endpoint"`. 

One possibility is that the following directory is empty: `Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint`. Could you double-check whether this directory is empty on your machine? If so, please run `git submodule update --init --recursive` to download this submodule again. Please let me know if the issue can be resolved by re-downloading the submodule?

Also, would you mind letting us know which operating system you are using? Thank you.

-Peifeng

NathanielRose, 2021-01-08:
@peifeng-unity That seemed to work, had an issue with my project setup hence the recursive submodule update was not working. Thanks!

closed 2021-01-08 by NathanielRose