Unity-Technologies/Unity-Robotics-Hub #102: Applying linear control to an articulation body in cartesian frame

karthikm-0, 2021-01-25:
I am wondering whether it is possible to move a specific articulation body (e.g. gripper) with respect to the base frame in x,y, and z coordinates. I would like to build such a controller to mimic a teleoperation mode provided by a robot but in Unity simulation (https://www.kinovarobotics.com/site...Ultra_lightweight_robot-User_guide_EN_R01.pdf, page 27) but the only variables the default controller allows me to manipulate are the joint angles of each individual joint. The default controller referenced here is this one: https://github.com/Unity-Technologi...pendix.md##Guide-to-write-your-own-controller).

mpinol, 2021-01-26:
Hey @karthikm-0 

It sounds like you want to be able to move the gripper's x,y,z coordinates in the Unity scene and have the robot's joints moved accordingly to accommodate the gripper's new position. Is this correct?

karthikm-0, 2021-01-26:
Yes, this is correct. I'd like the gripper to move in x, y, z with respect to the base frame. I also want to control the rotation with respect to its own frame. I assumed that modifying the linear and angular velocity of the gripper joint would suffice but I wasn't able to get it to move.

mpinol, 2021-01-27:
Hey @karthikm-0 

I believe to accomplish this you will need to implement inverse kinematics,
https://medium.com/unity3danimation/overview-of-inverse-kinematics-9769a43ba956

Sadly this is not something we provide 'out of the box'. However, if you are using ROS there are several IK solvers available as ROS packages.

If you decide to go the route of implementing your own IK then this link should be of help once you get to the maths portion,
https://docs.unity3d.com/2021.1/Documentation/ScriptReference/ArticulationBody.GetDenseJacobian.html

I hope this helps!

karthikm-0, 2021-02-01:
Thanks a lot! I decided to use a ROS package to send Twist messages to control my robot arm instead.

closed 2021-02-01 by karthikm-0