Unity-Technologies/Unity-Robotics-Hub #106: libvahcdl.dll issues and inconsistent environment behaviors

NathanielRose, 2021-01-31:
When I load the Part1Done scene I see that the niryo_one works and is fully functional as seen here:

[image]

Yet when I delete the niryo_one and import the same `niryo_one.urdf` file into the project I see this behavior when I play the scene (*No errors in the console*):

[image]

Additionally, I'm noticing this error when I attempt to import the urdf on my other windows machine. It was previously running successful but now fails due to an issue finding the vahcdl.dll

What could be the issues?

```
DllNotFoundException: libvhacd
MeshProcess.VHACD.GenerateConvexMeshes (UnityEngine.Mesh mesh) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Runtime/VHACD/VHACD.cs:138)
RosSharp.Urdf.Editor.UrdfGeometryCollision.ConvertMeshToColliders (UnityEngine.GameObject gameObject, System.Boolean setConvex) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfGeometryCollision.cs:131)
RosSharp.Urdf.Editor.UrdfGeometryCollision.CreateMeshCollider (RosSharp.Urdf.Link+Geometry+Mesh mesh) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfGeometryCollision.cs:71)
RosSharp.Urdf.Editor.UrdfGeometryCollision.Create (UnityEngine.Transform parent, RosSharp.Urdf.GeometryTypes geometryType, RosSharp.Urdf.Link+Geometry geometry) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfGeometryCollision.cs:44)
RosSharp.Urdf.Editor.UrdfCollisionExtensions.Create (UnityEngine.Transform parent, RosSharp.Urdf.Link+Collision collision) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfCollisionExtensions.cs:54)
RosSharp.Urdf.Editor.UrdfCollisionsExtensions.Create (UnityEngine.Transform parent, System.Collections.Generic.List`1[T] collisions) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfCollisionsExtensions.cs:35)
RosSharp.Urdf.Editor.UrdfLinkExtensions.Create (UnityEngine.Transform parent, RosSharp.Urdf.Link link, RosSharp.Urdf.Joint joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:27)
RosSharp.Urdf.Editor.UrdfLinkExtensions.ImportLinkData (RosSharp.Urdf.UrdfLink urdfLink, RosSharp.Urdf.Link link, RosSharp.Urdf.Joint joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:65)
RosSharp.Urdf.Editor.UrdfLinkExtensions.Create (UnityEngine.Transform parent, RosSharp.Urdf.Link link, RosSharp.Urdf.Joint joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:31)
RosSharp.Urdf.Editor.UrdfRobotExtensions.Create (System.String filename, RosSharp.ImportSettings settings) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfRobotExtensions.cs:74)
RosSharp.Urdf.Editor.FileImportMenu.OnGUI () (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/MenuItems/FileImportMenu.cs:58)
UnityEditor.HostView.InvokeOnGUI (UnityEngine.Rect onGUIPosition, UnityEngine.Rect viewRect) (at <27ac941b3da34983800e9a41b73eceef>:0)
UnityEditor.DockArea.DrawView (UnityEngine.Rect viewRect, UnityEngine.Rect dockAreaRect) (at <27ac941b3da34983800e9a41b73eceef>:0)
UnityEditor.DockArea.OldOnGUI () (at <27ac941b3da34983800e9a41b73eceef>:0)
UnityEngine.UIElements.IMGUIContainer.DoOnGUI (UnityEngine.Event evt, UnityEngine.Matrix4x4 parentTransform, UnityEngine.Rect clippingRect, System.Boolean isComputingLayout, UnityEngine.Rect layoutSize, System.Action onGUIHandler, System.Boolean canAffectFocus) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, UnityEngine.Matrix4x4 worldTransform, UnityEngine.Rect clippingRect, System.Action onGUIHandler, System.Boolean canAffectFocus) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, System.Action onGUIHandler, System.Boolean canAffectFocus) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, System.Boolean canAffectFocus) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.IMGUIContainer.SendEventToIMGUIRaw (UnityEngine.UIElements.EventBase evt, System.Boolean canAffectFocus, System.Boolean verifyBounds) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.IMGUIContainer.SendEventToIMGUI (UnityEngine.UIElements.EventBase evt, System.Boolean canAffectFocus, System.Boolean verifyBounds) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.IMGUIContainer.HandleEvent (UnityEngine.UIElements.EventBase evt) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.CallbackEventHandler.HandleEventAtTargetPhase (UnityEngine.UIElements.EventBase evt) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.MouseCaptureDispatchingStrategy.DispatchEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcher.ApplyDispatchingStrategies (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, System.Boolean imguiEventIsInitiallyUsed) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcher.ProcessEventQueue () (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcher.OpenGate () (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcherGate.Dispose () (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.EventDispatcher.Dispatch (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, UnityEngine.UIElements.DispatchMode dispatchMode) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.BaseVisualElementPanel.SendEvent (UnityEngine.UIElements.EventBase e, UnityEngine.UIElements.DispatchMode dispatchMode) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.UIElementsUtility.UnityEngine.UIElements.IUIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr, System.Boolean& eventHandled) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.UIEventRegistration.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.UIElements.UIEventRegistration+<>c.<.cctor>b__1_2 (System.Int32 i, System.IntPtr ptr) (at <519636828bf44b44aa4c317f8b947afe>:0)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr, System.Boolean& result) (at <931b26b61f444e3485195721b0227b20>:0)
```

at669, 2021-02-01:
Hi @NathanielRose, 

> Yet when I delete the niryo_one and import the same niryo_one.urdf file into the project I see this behavior when I play the scene (No errors in the console):

This is expected behavior. In the [Unity Setup](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/pick_and_place/1_urdf.md#setting-up-the-robot) part of the tutorial, the base of the robot is marked `Immovable`. Upon reimporting the robot, you will have to re-check the newly imported base_link to also be immovable to stop it from falling over.

> Additionally, I'm noticing this error when I attempt to import the urdf on my other windows machine. It was previously running successful but now fails due to an issue finding the vahcdl.dll

Thanks for reporting this issue! So that we can look into this further:

1. You mentioned it the importer was previously successful. Was this also on the same Windows machine, or on the first device you used the importer?
2. What version of the URDF-Importer is in your project?  You can find this in the Package Manager window or in Packages/packages-lock.json. 
3. What version of Unity does this happen in?
4. What is your build target (e.g. x86 or x86_64)?

NathanielRose, 2021-02-01:
Thanks for the response @at669 ! The `Immovable` config fixed it.

> You mentioned it the importer was previously successful. Was this also on the same Windows machine, or on the first device you used the importer?

On the same machine, but I did install ROS Melodic on my Windows subsystem for Linux. unsure if that had any effect.

> What version of the URDF-Importer is in your project? You can find this in the Package Manager window or in Packages/packages-lock.json.

The URDF-Importer is version: `0.1.2-preview`

> What version of Unity does this happen in?

Unity Version: `2020;2;0b9`

> What is your build target (e.g. x86 or x86_64)?

I believe its x86_64. Is there a way to check?
Is there a substantial difference between using the Unity option for the mesh decomposer instead of the VHACD option?

at669, 2021-02-01:
> I believe its x86_64. Is there a way to check?

Thanks! To verify, in Unity's top menu bar, it's found under File > Build Settings, and it should be listed under Target Platform/Architecture in the Build Settings window.

> Is there a substantial difference between using the Unity option for the mesh decomposer instead of the VHACD option?

There is more information about VHACD [here](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md#convex-mesh-collider). Ultimately, the Unity mesh decomposer may have worse performance in simulation, but it is a temporary workaround as we investigate the DllNotFoundException.

NathanielRose, 2021-02-01:
Confirming that the Target Platform: `x86_64`
Ok great, I will work with Unity for mesh decomposer until the DLL issue is unblocked.

m-talha, 2021-02-02:
+1 on the libvahc.dll error when following the [tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/1_urdf.md). My system specs:

Unity 2020.2.0b9
URDF-Importer: 0.0.4-preview
Platform: Win 10 x86_64

at669, 2021-02-02:
Hi @NathanielRose and @m-talha,

I can't seem to reproduce the error on my machine. Could you try:

1. Moving the `com.unity.robotics.urdf-importer@...` folder from `.../PickAndPlaceProject/Library/PackageCache` to `.../PickAndPlaceProject/Packages` to make the files editable,
2. In Unity, navigating to `Packages/com.unity.robotics.urdf-importer/Runtime/VHACD/libvhacd.dll`, and checking x86 (and applying the change), and finally,
[image]

3. In Unity, navigating to `Packages/com.unity.robotics.urdf-importer/Runtime/Unity.Robotics.URDFImporter.asmdef` and checking Windows 32-bit (and applying the change),
[image]

and try to re-import the URDF using VHACD?

m-talha, 2021-02-02:
Thanks @at669 for the suggestions but the problem is still there.

at669, 2021-02-03:
Thanks for letting me know, @m-talha. We will continue to look into this issue.

LaurieCheers-unity, 2021-02-04:
Hi, we believe the problem is that the VHACD library we're using depends on a dll called vcomp140d.dll. As a workaround while we figure out a better solution, I've attached the file; unzip it and put the dll in your C:/Windows/System32 folder.
[vcomp140d.zip](https://github.com/Unity-Technologies/Unity-Robotics-Hub/files/5928525/vcomp140d.zip)

macg30, 2021-02-07:
Hi, I had the same problem like @NathanielRose and @m-talha, but I have added the vcomp140d.dll in that system path C:/Windows/System32 ( as @LaurieCheers-unity indicates) and the URDF importer with VHACD option worked fine ;), although the wait is more than the other option.

Thank you for the support! @at669 and @LaurieCheers-unity 
[image]

NathanielRose, 2021-02-09:
This worked for me as well! Unsure if we want to close the ticket or wait until a fix in the repo is provided. I'm fine with either!

m-talha, 2021-02-09:
+1 Can also confirm it works

closed 2021-02-10 by LaurieCheers-unity