Unity-Technologies/Unity-Robotics-Hub #108: ROS servers with Unity and WSL

NathanielRose, 2021-02-02:
Hello,

I am running into an issue where I have configured the `robotics_demo` `server_endpoint` and am running the RosPublisher from Unity but receive a `TCPConnecectorException` stating: `No connection could be made because the target machine actively refused it`. The ROS Settings have been correctly configured.

Is this an issue with connecting to ROS running in WSL?

[image]

at669, 2021-02-02:
Hi @NathanielRose,

This does look like an issue with the WSL connection! 

First, I would suggest verifying the virtual network interface's IP via running `ifconfig` in the WSL environment and checking `inet`. This should be the address used as your `ROS_IP` value as your rosparam (or in your config/params.yaml, depending on how you've configured this setup), and as the ROS IP Address in your ROS Settings.

If that's already the case, you may also need to configure your firewall to allow traffic between WSL and Windows, adding both Inbound and Outbound rules for WSL traffic.

Finally, if the above don't work, I would try adding the Windows-side virtual address as the Override Unity IP Address in the ROS Settings. 

Let me know if any of these methods resolve your issue!

NathanielRose, 2021-02-02:
Hi @at669 thanks for the support here!

I adjusted the ROS IP address based on the `inet` address and still receive the same behavior.

[image]

I received the same results when I turned off my windows firewall completely.

Where would I identify the windows-side virtual address?

at669, 2021-02-02:
You can find the address running `ipconfig` on the Windows side! 

However, it looks like the server endpoint is still running with the old address--you'll want to do a `rosparam load ./config/params.yaml` before your `rosrun robotics_demo server_endpoint.py`. I would try this before adding the Override Unity IP, but the network settings may vary due to your setup, so you may have to do a bit of debugging and trying the override even after reloading your rosparams. Let me know how it goes!

NathanielRose, 2021-02-03:
@at669 This worked! I had to run the `osparam load ./config/params.yaml` beforehand.

I went a bit ahead and was attempting to customize ROS for a different robot.
I am modeling what the team did in the cube ROSpublish tutorial, publishing my robots position & rotations to the TCP endpoint. Unsure if it makes sense to open another ticket but I was running into issue:

I am currently able to write my positions to the Unity console, but unable to publish on the ROS server.
[image]

When I start my scene I receive this error:
```
NullReferenceException: Object reference not set to an instance of an object
UR10TrajectoryPlanner.Update () (at Assets/Scripts/UR10TrajectoryPlanner.cs:165)
```
My `UR10MoveItJoints` are modeled after the `NiryoMoveItJoints`.
Any idea why I am getting a `NullReferenceException`?

at669, 2021-02-03:
Glad to hear the rosparams worked!

> Any idea why I am getting a NullReferenceException?

My guess would be that the `ROSConnection ros` variable hasn't been set by the time you call `ros.Send()`. Since it looks like you've used the ROS Settings workflow, you likely have an `Assets/Resources/ROSConnectionPrefab.prefab` file that contains your ROS settings. You can refer to this by assigning `ros = ROSConnection.instance;` in your TrajectoryPlanner's `Start()` or `Awake()` functions, which should hopefully resolve the NullReferenceException.

NathanielRose, 2021-02-10:
This worked, Thanks for the help here @at669 !

closed 2021-02-10 by NathanielRose