Unity-Technologies/Unity-Robotics-Hub #115: Collision detection for trajectory planning with moveit?

WenqiangX, 2021-02-03:
Hi guys,

According to the pick-and-place tutorial, it seems the unity side just send the pose to the moveit, so the moveit cannot perform the collision detection, right?

Correct me if I miss something, thanks!

WenqiangX, 2021-02-03:
I have encounter some internet issue, sorry for the duplicated posts.

at669, 2021-02-03:
Hi @WenqiangX,

MoveIt will calculate the robot's trajectories to avoid itself, as it is aware of the URDF description loaded during [launch](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/pick_and_place/moveit_file_descriptions.md). If you would like to add objects to the scene for collision avoidance, such as the table or other objects in the scene, you can do so easily with the [Move Group Interface](http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#adding-objects-to-the-planning-scene) with calls to `add_box`, for example. 

If you plan on passing the position of objects from Unity to ROS, please also note a conversion between Unity and ROS coordinates will need to take place, which can be done using the [ROSGeometry](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/main/ROSGeometry.md) component.

Let me know if that answers your question!

WenqiangX, 2021-02-03:
Thanks for your reply, @at669 

Yes I would like to add objects to the scene for collision avoidance. So if I understand correctly, to handle it, I should also import the models from the MoveIt side (Move Group Interface). In other words, a set of models (the table or other objects) for display in Unity, and a separate set of models (the table or other objects) for collision avoidance in MoveIt? About twice the consumption of the memory to store the object models, right?

I thought a more light-weighted approach exists for passing the collision model from Unity to ROS, since Unity has its own collision detection between the robot and the object.

at669, 2021-02-03:
It will depend on your specific use case--if the objects you'd like to add to the scene are primitives, you can use MoveIt's `add_box`, `add_sphere`, or `add_plane`, and these object meshes can live on the Unity side and only need to pass position and scale parameters to MoveIt. 

If you have more complicated shapes that cannot be approximated using primitives, you can look into sending the [Mesh](https://docs.unity3d.com/ScriptReference/Mesh.html) from Unity to MoveIt as a [CollisionObject](http://docs.ros.org/en/melodic/api/moveit_msgs/html/msg/CollisionObject.html), and using calls to [`addCollisionObjects`](http://docs.ros.org/en/noetic/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1PlanningSceneInterface.html#ab1ef88c061ed75784d34d1a1bbe6d7a5) to the planning scene.

These are options if you would like to use MoveIt to avoid collisions during motion planning--if you'd only like to track collisions when they occur using Unity's built-in collision detection, you can do so with tracking [OnCollisionEnter](https://docs.unity3d.com/2020.2/Documentation/ScriptReference/Collider.OnCollisionEnter.html) calls on the desired object colliders.

WenqiangX, 2021-02-03:
Thanks, your answer fully addressed my concerns!

at669, 2021-02-03:
Glad to answer your questions! I will close out this issue for now--feel free to open it up again if you have any follow-up issues or questions.

closed 2021-02-03 by at669