Unity-Technologies/Unity-Robotics-Hub #125: Inconsistent behavior of Articulation Bodies when starting scene

karthikm-0, 2021-02-08:
Hi,

I'm trying to launch my robot and subscribe to a joint states topic from ROS. I find that the robot is extremely unstable at times and does not reach the start state of the ROS robot. Other times, this works perfectly. I've uploaded a short gif of a few runs where the first few are fine (https://media.giphy.com/media/ePnzMhBtlsfA5d1pRy/giphy.gif).  Sometimes the robot keeps moving to different positions in a rapid manner (https://media.giphy.com/media/ePnzMhBtlsfA5d1pRy/giphy.gif). Could someone help me troubleshoot the issue?

Thanks!

vidurvij-Unity, 2021-02-08:
Hi @karthikm-0.
Thanks for reaching out with the problem. Is it possible to get access to your repo or a zip file of the project. I would be able to help you more if I know the in more detail about your project.
Vidur

karthikm-0, 2021-02-09:
Thanks @vidurvij-Unity! I've added you to my private repo. You'll need to import the tcp connector package before running. Also you'll need to run the program a few times to see the weird behavior (sometimes the gripper joints are oddly positioned and other times it goes crazy). Also, on a side note, I've been trying to import a specific robot (Franka Panda) and keep running into the AA BB error. You can find the urdf and meshes here: https://github.com/justagist/franka_panda_description.

vidurvij-Unity, 2021-02-09:
@karthikm-0 I got your invitation. I don't see any instructions to replicate the issue. Can you update the project readme so I can replicate the issue. Also I would recommend using the package manager to use the ROS-TCP-Connector to use the package.
Thanks 
Vidur

karthikm-0, 2021-02-18:
Hi @vidurvij-Unity sorry for the late response. I've decided to abandon this approach for now and am trying to import the Franka Panda using this URDF https://github.com/justagist/franka_panda_description but I am seeing the aa bb issue yet again. Could you try to import this robot and see whether it works? Thanks!

karthikm-0, 2021-02-19:
@vidurvij-Unity I just realized I may have posted this in the wrong repo as it may be better suited in urdf-importer (https://github.com/Unity-Technologies/URDF-Importer/issues/25). Should I close this and open a post there? Thanks!

vidurvij-Unity, 2021-02-19:
@karthikm-0 Does this issue appear when you press "play" ?

karthikm-0, 2021-02-19:
@vidurvij-Unity Yes, the issue appears when I hit play. Please see the attached gif:

https://gph.is/g/4zAbWw0

This occurs despite ignoring collisions between all links and disabling physics for the robot through layers.

karthikm-0, 2021-02-20:
@vidurvij-Unity Quick update - I've isolated the issue as belonging to a specific articulation body - panda_link8.

vidurvij-Unity, 2021-02-22:
@karthikm-0 Thank you bringing this to our attention. I will resolve this bug.

closed 2021-03-09 by karthikm-0