Unity-Technologies/Unity-Robotics-Hub #149: combining Unity-Robotics with ML-agents

tlpss, 2021-02-15:
Hi

I would like to combine the functionalities of the URDF importer and TCP Connector with the [ml-agents](https://github.com/Unity-Technologies/ml-agents)  framework, to mimic a real-world setup with ROS while still making use of the features provided by ml-agents for interfacing with the environment in python such as the gym-interface and the side-channels.

However at first sight I cannot find any guidelines on how to do this. Is there any information available on how to combine the functionalities of these two packages?

hyounesy, 2021-02-18:
Hi, Unfortunately we currently don't have an official guideline or a comprehensive tutorial released for using the ml-agents together with the robotics framework, although that is something that we do have in mind for future.
In the meantime you can take a look at the following tutorial on training a robot arm with ML-agents:
https://github.com/Unity-Technologies/articulations-robot-demo/tree/mlagents
Hopefully that helps you to get started with your project. Please let us know if you have any questions or issues and we will be glad to help.

mpinol, 2021-03-15:
Closing due to inactivity. Please don't hesitate to reach out if you have any other questions!

closed 2021-03-15 by mpinol