Unity-Technologies/Unity-Robotics-Hub #153: Missing "Robotics" menu in the editor

m-talha, 2021-02-16:
Hi,

I seem to be missing the "Robotics" menu in the Unity editor. Instead my menu looks like this:

[image]

Following the tutorial for pick and place, I need the menu for the ROS settings. Is the tutorial outdated or is there something wrong with my installation?

Thanks

hyounesy, 2021-02-18:
Hi @m-talha, we just recently updated our ROS-TCP-Packages and the tutorial. So you may be using an older version of the project with the newer documentation. Could you please try getting the most recent version of the Pick and Place project?

m-talha, 2021-02-25:
Updating the package solved it thanks.

closed 2021-02-25 by m-talha