Unity-Technologies/Unity-Robotics-Hub #156: Pick and place demo: robot not moving in part 3

KKassem, 2021-02-17:
Hello.

I have managed to replicate the steps all the way from part 0 till part 3. The last step would be to press the publish button. Clicking on this button triggers multiple motion planning/trajectory calculation requests on the ROS side. The robot in unity just stays perfectly still. I noticed after some time the gripper opens. 

I'm running ROS noetic on WSL for windows 10 with Ubuntu 20.04. I have two versions of Unity installed: 2020.2.4f1 as well as 2020.2.0b9, with the tutorial running on the latter.

ROS-side console  output : 

 
```
[ INFO] [1613574625.397392500]: Loading robot model 'niryo_one'...
[ INFO] [1613574625.398079400]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1613574625.705789400]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1613574626.809087400]: Ready to take commands for planning group arm.
[ INFO] [1613574627.035894900]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ WARN] [1613574627.036133900]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[ INFO] [1613574627.036223600]: Planning attempt 1 of at most 1
[ INFO] [1613574627.038109100]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574627.038721700]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574627.049564200]: arm/arm: Created 4 states (2 start + 2 goal)
[ INFO] [1613574627.049646100]: Solution found in 0.011161 seconds
[ INFO] [1613574627.052300400]: SimpleSetup: Path simplification took 0.001223 seconds and changed from 3 to 2 states
[ INFO] [1613574627.054228800]: Disabling trajectory recording
[ INFO] [1613574635.455153100]: Received trajectory RESULT
[ INFO] [1613574635.455391600]: Controller niryo_one_follow_joint_trajectory_controller successfully finished
[ INFO] [1613574635.455560300]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1613574635.475760800]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1613574635.476896200]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574635.477080700]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574635.487536000]: arm/arm: Created 4 states (2 start + 2 goal)
[ INFO] [1613574635.487615800]: Solution found in 0.010578 seconds
[ INFO] [1613574635.490486900]: SimpleSetup: Path simplification took 0.002458 seconds and changed from 3 to 2 states
[ INFO] [1613574635.515298900]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ WARN] [1613574635.515413000]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[ INFO] [1613574635.515471800]: Planning attempt 1 of at most 1
[ INFO] [1613574635.516218700]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574635.516379900]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574635.526968700]: arm/arm: Created 4 states (2 start + 2 goal)
[ INFO] [1613574635.527057100]: Solution found in 0.010711 seconds
[ INFO] [1613574635.528996500]: SimpleSetup: Path simplification took 0.001051 seconds and changed from 3 to 2 states
[ INFO] [1613574637.929882100]: Received trajectory RESULT
[ INFO] [1613574637.930037700]: Controller niryo_one_follow_joint_trajectory_controller successfully finished
[ INFO] [1613574637.930146600]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1613574637.955422600]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1613574637.956379000]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574637.956764300]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574637.977600100]: arm/arm: Created 5 states (2 start + 3 goal)
[ INFO] [1613574637.977691400]: Solution found in 0.021039 seconds
[ INFO] [1613574637.978974000]: SimpleSetup: Path simplification took 0.001227 seconds and changed from 4 to 2 states
[ INFO] [1613574638.015486400]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ WARN] [1613574638.015629600]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[ INFO] [1613574638.015697800]: Planning attempt 1 of at most 1
[ INFO] [1613574638.016609500]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574638.016827600]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574638.027434000]: arm/arm: Created 4 states (2 start + 2 goal)
[ INFO] [1613574638.027548600]: Solution found in 0.010770 seconds
[ INFO] [1613574638.029289100]: SimpleSetup: Path simplification took 0.001298 seconds and changed from 3 to 2 states
[ INFO] [1613574640.430118800]: Received trajectory RESULT
[ INFO] [1613574640.430311900]: Controller niryo_one_follow_joint_trajectory_controller successfully finished
[ INFO] [1613574640.430411800]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1613574640.455818800]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1613574640.456663700]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574640.456821800]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574640.467361300]: arm/arm: Created 4 states (2 start + 2 goal)
[ INFO] [1613574640.467434000]: Solution found in 0.010641 seconds
[ INFO] [1613574640.469247100]: SimpleSetup: Path simplification took 0.001671 seconds and changed from 3 to 2 states
[ INFO] [1613574640.495590200]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ WARN] [1613574640.495711400]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[ INFO] [1613574640.495771400]: Planning attempt 1 of at most 1
[ INFO] [1613574640.496638800]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574640.496840900]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574640.507447200]: arm/arm: Created 5 states (2 start + 3 goal)
[ INFO] [1613574640.507536800]: Solution found in 0.010743 seconds
[ INFO] [1613574640.510536600]: SimpleSetup: Path simplification took 0.002870 seconds and changed from 4 to 2 states
[ INFO] [1613574650.662529200]: Received trajectory RESULT
[ INFO] [1613574650.662774300]: Controller niryo_one_follow_joint_trajectory_controller successfully finished
[ INFO] [1613574650.662884900]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1613574650.675734400]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1613574650.676551500]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1613574650.676733500]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1613574650.687357000]: arm/arm: Created 5 states (2 start + 3 goal)
[ INFO] [1613574650.687433100]: Solution found in 0.010743 seconds
[ INFO] [1613574650.690867300]: SimpleSetup: Path simplification took 0.003085 seconds and changed from 4 to 2 states
```

I would appreciate any help with tracking down the source of the issue on this. Thanks.

hyounesy, 2021-02-18:
Hi, I don't see any issues in the ROS console output.  Do you see any errors on the Unity Editor console?
Looks like the message from Unity was sent to MoveIt without issue, but MoveIt failed to send a message back to Unity.
One possibility is that a firewall is blocking some connections from the WSL environment. Are you able to ping your local machine's IP address from WSL? If not please try [adding a firewall rule](http://lexisnexis.custhelp.com/app/answers/answer_view/a_id/1081611/~/adding-exceptions-to-the-windows-firewall) to allow incoming connections from your WSL to your local machine?

KKassem, 2021-02-18:
Thanks for responding. 

You're right. I had a feeling this might have been a firewall issue. Ping wasn't possible, and so I tried temporarily disabling the firewall on Windows for the test, and the ping managed to go through. However, the arm still doesn't move.

Checking the Unity console, it has an error reading 

```
No data available on network stream after 10 attempts.
UnityEngine.Debug:LogError (object)
ROSConnection/<SendServiceMessage>d__21`1<RosMessageTypes.NiryoMoveit.MoverServiceResponse>:MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@ab45157ad3/Runtime/TcpConnector/ROSConnection.cs:129)
UnityEngine.UnitySynchronizationContext:ExecuteTasks ()
```

This error shows with the firewall turned off. Pings go through normally. I checked the TCP connection by running telnet on port 5005, which also seems to be running normally.

hyounesy, 2021-02-19:
Here are a couple things to investigate:
- What IP address did you set in the unity address override? It might be because you are not using the correct IP addresses for this.  I believe in the WSL that has to be set manually.
- See if the solution proposed [for this similar issue](https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/108) works for you. In short: Use ifconfig inside WSL to find the address of the eth0 adapter, use ipconfig on the windows side to find the corresponding vEthernet WSL adapter and make sure those values are what are used wherever IP addresses are needed.

Let us know if any of these methods resolve your issue!

mrceresa, 2021-03-02:
Did you check if it is the timeout? 

I had the same error when my machine took 11 seconds to calculate the trajectory. I put it to 20 secs and it worked. You can change it in the ROS Settings.

hyounesy, 2021-03-12:
Hello @KKassem , just wanted to check if you were able to resolve the issue above, before we close the issue.

KKassem, 2021-03-13:
> Did you check if it is the timeout?
> 
> I had the same error when my machine took 11 seconds to calculate the trajectory. I put it to 20 attempts and it worked. You can change it in the ROS Settings.

It wasn't a time out afaik because the gripper opens at the end.. changing the number of attempts did nothing.

It was too frustrating to deal with, so I think I will try further with the Docker approach. The WSL path just doesn't seem robust enough at the moment if it's this undeterministic. Thanks for all your help.

peifeng-unity, 2021-03-24:
Thank you @KKassem for your feedback. Hope everything would go well with the Docker approach.

I will close this issue. Please feel free to reopen it if you have more questions.

closed 2021-03-24 by peifeng-unity