Unity-Technologies/Unity-Robotics-Hub #160: Encounter error with ROS-TCP-Connector

KKKKKyle, 2021-02-20:
Hi, thanks for helping in advance.

I encountered this error while I was adding the ROS-TCP-Connector through Unity's Package Manager:

[image]

Must I have ROS installed first before import this package? Also, may it be a platform-related question? I am currently trying this on my mac.

grahamday266, 2021-02-21:
I too am having this issue on a 20.04 Ubuntu installation running Noetic ROS.  My Unity version is 2020.2.5f1

Error message for 1 of 3 is:

[Package Manager Window] Unable to add package [https://github.com/Unity-Technologies/ROS-TCP-Connector.git]:
  [https://github.com/Unity-Technologies/ROS-TCP-Connector.git] does not point to a valid package repository. No package manifest was found. Verify the repository URL and make sure the package is located in the root folder of the repository.
UnityEditor.EditorApplication:Internal_CallUpdateFunctions () (at /home/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:327)

grahamday266, 2021-02-21:
I was able to import by cloning the Git library to a local folder and then adding the package via the "Add package from disk..." option.  You just have to find the package.json file which is in the com.unity.robotics.ros-tcp-connector folder.

KKKKKyle, 2021-02-21:
> I was able to import by cloning the Git library to a local folder and then adding the package via the "Add package from disk..." option. You just have to find the package.json file which is in the com.unity.robotics.ros-tcp-connector folder.

I believe we encountered a different issue. The error report I had here appeared after I picked the package.json file.

peifeng-unity, 2021-02-23:
Hi @KKKKKyle, it is great to hear your interest to try the ROS connectors. For your questions, the ROS-TCP-Connector is an independent package from the ROS environment, so you don't have to install ROS before importing the package. We also have tested importing the ROS-TCP-Connector via Package Manager on macOS, and it should work fine. 

Would you mind letting us know how you are adding the package through Unity's Package Manager? It would be helpful to identify what went wrong on your mac.

The recommended way to add ROS-TCP-Connector through the Package Manager can be found in [this tutorial (ROS-Unity Initial Setup/Unity Scene)](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#unity-scene).

peifeng-unity, 2021-02-23:
Hi @grahamday266, thank you for your interest to try the ROS connector too. We have updated the path of the Unity package in the ROS-TCP-Connector repository, and the git URL needs to specify the path after the git repository link, e.g. https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector. A more detailed instruction can be found on [this page](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#unity-scene).

Please let me know if it still doesn't work for you and we could help to debug and fix the issue.

KKKKKyle, 2021-02-23:
Hi @peifeng-unity 

I previously tried both installing from local and from git url. I just tried again to install it by following the tutorial you shared (1. Create a new scene, 2. use the link to install from git url in Unity's Package Manager), but the same issue appears again.

Here are the two errors:

> Library/PackageCache/com.unity.robotics.ros-tcp-connector@2b09bf7e77/Runtime/ROSGeometry/ROSVector3.cs(200,104): error CS1501: No overload for method 'ToString' takes 2 arguments
> 
> Library/PackageCache/com.unity.robotics.ros-tcp-connector@2b09bf7e77/Runtime/ROSGeometry/ROSQuaternion.cs(143,95): error CS1501: No overload for method 'ToString' takes 2 arguments

Here are the two warnings:

> Library/PackageCache/com.unity.robotics.ros-tcp-connector@2b09bf7e77/Runtime/TcpConnector/ROSConnection.cs(152,24): warning CS0168: The variable 'serviceName' is declared but never used
> 
> Library/PackageCache/com.unity.robotics.ros-tcp-connector@2b09bf7e77/Runtime/TcpConnector/HUDPanel.cs(18,14): warning CS0414: The field 'HUDPanel.redrawGUI' is assigned but its value is never used

I'm very new to Unity and C# but I'm guessing it may relate to the compiler for C#?  The Unity version I'm using is 2019.4.15f1.

peifeng-unity, 2021-02-23:
Hi @KKKKKyle,

Thanks for the information. The issue should be caused by Unity's version. 

In Unity 2019.4, the `UnityEngine.Vector.ToString()` only takes zero or one argument (see [API doc here](https://docs.unity3d.com/2019.4/Documentation/ScriptReference/Vector3.ToString.html)). Starting from Unity 2020.2, the `UnityEngine.Vector.ToString()` begins to support two arguments (see [API doc here](https://docs.unity3d.com/2020.1/Documentation/ScriptReference/Vector3.ToString.html)).

In this case, you may need to use 2020.1 or a newer version of Unity. Please let me know if the issue could be resolved in this way.

KKKKKyle, 2021-02-23:
Thank you peifeng. That did solve my issue. Thanks!

closed 2021-02-23 by KKKKKyle