Unity-Technologies/Unity-Robotics-Hub #161: Can't manually add the packages

ZahraaBass, 2021-02-22:
Hello, 
when I'm trying to add the TCP connector package from window/packagemanager and add package from git url , I get the following error:
[Package Manager Window] Error adding package: https://github.com/Unity-Technologies/ROS-TCP-Connector.git.
UnityEditor.EditorApplication:Internal_CallUpdateFunctions () (at /Users/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:327)

I'm using Unity 2020.2.0f1

peifeng-unity, 2021-02-23:
Hi @ZahraaBass, it is great to hear your interest to try the ROS connector too. We have updated the path of the Unity package in the ROS-TCP-Connector repository, and the git URL needs to specify the path after the git repository link, e.g. https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector. A more detailed instruction can be found on [this page](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#unity-scene).

Please let me know if it still doesn't work for you and we could help to debug and fix the issue.

ZahraaBass, 2021-02-23:
@peifeng-unity It worked, thank you!

closed 2021-02-23 by ZahraaBass