Unity-Technologies/Unity-Robotics-Hub #164: MQTT

fahrulmr, 2021-02-23:
hi,
can i use this to combine on MQTT, the mechanism same like MQTT (publisher and subscriber)
and how to do that in here?

mpinol, 2021-02-24:
Hey @fahrulmr 

We currently do not support MQTT but it is something we have briefly talked about in the past.
Would you be willing to describe your use case? 

Also, I did a Google search and found several example projects available that seem to have integrated MQTT and Unity. It might be best to take a look at one of those for how to use MQTT with Unity.

mpinol, 2021-03-15:
Hey @fahrulmr , I am going to close this issue but if you have any other questions please don't hesitate to reach out again!

closed 2021-03-15 by mpinol