Unity-Technologies/Unity-Robotics-Hub #165: getting error when I Installing the Unity Robotics packages

KeyG518, 2021-02-24:
I got a error when I add the package from the package manager(quick setup part).

 [Package Manager Window] Unable to add package [https://github.com/Unity-Technologies/ROS-TCP-Connector.git]:
  [https://github.com/Unity-Technologies/ROS-TCP-Connector.git] does not point to a valid package repository. No package manifest was found. Verify the repository URL and make sure the package is located in the root folder of the repository.
UnityEditor.EditorApplication:Internal_CallUpdateFunctions ()

how should I solve this?

Thanks in advance!

peifeng-unity, 2021-02-24:
Hi @KeyG518 , it is great to hear your interest. Sorry for the confusion in the tutorial. Please refer to #161 for this issue.

We have updated the `ROS-TCP-Connector` repository and you need to add the path information to import the package, e.g. https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector. We will update the tutorial to fix this issue soon.

peifeng-unity, 2021-03-12:
Closing this issue since there is no more update. Please feel free to reopen it if necessary.

closed 2021-03-12 by peifeng-unity