Unity-Technologies/Unity-Robotics-Hub #169: Problems with Android build

Exo-dar, 2021-02-27:
My OS is: Windows 10 
My Unity Version is: Unity 2020.2.2f1
My ROS Distribution is: ROS Melodic Morenia
My Build Target Platform is: Android

Hi, I'm trying to build a project using Unity Robotics Hub in Android but it's not working.

I have no problems while connecting the Unity Editor with my ROS machine, but once I build it on Android it doesn´t work. The log says something related with serialization of the messages.

Looking for this issue I found something related in the ROS# package. They say it's the way the newtonsoft.Json interacts with the IL2CPP scripting backend, which is mandatory for Android. They solve this issue by changing the package which handles the serialization and the protocol. This is the topic where they solve it: https://github.com/siemens/ros-sharp/issues/287.

Is there a way to do some workaround or something in order to achieve this with Unity Robotics Hub? Or I should migrate my project to their package if I want to build an Android apk?

Thanks in advance!

LaurieCheers-unity, 2021-03-01:
Hi, I'm afraid the project is not currently set up to easily build for Android, but it should be possible. You'll need to tick the "Android" checkbox in the file Runtime/Unity.Robotics.ROSTCPConnector.asmdef in the ROS TCP Connector package.

Unfortunately Unity makes it hard to edit the contents of a package. The easiest approach is probably to copy the whole package into your project's Assets folder. Here's a process that works for me:
    1) In Unity, right click on the package ROS TCP Connector, and select "Show in Explorer" ("Reveal in Finder" on MacOS). This should open an OS folder window.
    2) In the OS window, copy the folder com.robotics.ros-tcp-connector@whatever into a safe place, say your Documents folder.
    3) Back in Unity, open Window/PackageManager and select the ROS TCP Connector package. Click the "Remove" button at the bottom right to uninstall the package.
    4) In Unity, create a folder called "RosTcpConnector" somewhere in your Assets folder. Right click on it and select "Show in Explorer". Then close Unity.
    5) Copy the contents of the com.robotics.ros-tcp-connector folder you saved earlier into the RosTcpConnector folder.

Now you should be able to reopen Unity, go into the RosTcpConnector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef file and tick the "Android" checkbox so that it will be included in Android builds.

If your project is also using URDF-Importer, you'll probably have to do the same for that package. Hope that helps! I'll look into getting the Android checkbox ticked on these files by default.

mpinol, 2021-03-15:
Hey @Exo-dar, were you able to resolve your issue?

Exo-dar, 2021-03-16:
Yep, @LaurieCheers-unity's answer did solve the problem.

Thank you so much :D

closed 2021-03-16 by mpinol

geoffreymantel, 2022-01-03:
@mpinol  - Shouldn't Android work out of the box? i.e:

```
diff --git a/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef b/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef
index ca790bf..dfaed54 100644
--- a/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef
+++ b/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef
@@ -8,7 +8,8 @@
         "macOSStandalone",
         "WSA",
         "WindowsStandalone32",
-        "WindowsStandalone64"
+        "WindowsStandalone64",
+        "Android"
     ],
     "excludePlatforms": [],
     "allowUnsafeCode": false,
```

I also just encountered this issue when building for Oculus Quest 2. The work-around here seems non-trivial compared with the suggested patch above. Please let me know what you think.

ZuWang95, 2024-05-22:
> Hi, I'm afraid the project is not currently set up to easily build for Android, but it should be possible. You'll need to tick the "Android" checkbox in the file Runtime/Unity.Robotics.ROSTCPConnector.asmdef in the ROS TCP Connector package.
> 
> Unfortunately Unity makes it hard to edit the contents of a package. The easiest approach is probably to copy the whole package into your project's Packages folder. Here's a process that works for me: 1) In Unity, right click on the package ROS TCP Connector, and select "Show in Explorer" ("Reveal in Finder" on MacOS). This should open an OS folder window called PackageCache. 2) Do the same for your project's Assets folder. This should open your project's root folder. 3) Within the root folder window, open the Packages folder. 4) Copy the folder com.robotics.ros-tcp-connector@whatever from your PackageCache window into the Packages folder.
> 
> Now, in Unity, you should be able to go into the RosTcpConnector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef file and tick the "Android" checkbox so that it will be included in Android builds.
> 
> If your project is also using URDF-Importer, you'll probably have to do the same for the URDF-Importer package. Hope that helps! I'll look into getting the Android checkbox ticked on these files by default.

Thanks for the help!