Unity-Technologies/Unity-Robotics-Hub #170: ROS TCP handshake fails in Part 2, even after double checking the ROS_IP settings

engineering-coder, 2021-02-28:
Just as stated in the Troubleshooting Section, "ROS-Unity server listening..." printed on the Unity side but no "ROS-Unity Handshake received" on the ROS side.
I am using Docker and I have set the ROS Settings in Unity as stated in the tutorial, and I also have set the "ROS_IP" into the correct one. I wonder what is going wrong?
Besides, I am using Unity 2020.2.3f1c1 Personal
[image]
[image]

peifeng-unity, 2021-03-01:
Hello @engineering-coder, it is good to hear your interest in using docker and trying the ROS connection. Sorry that the troubleshooting might be confusing here. 

You are doing nothing wrong and it should work well even though it doesn't print "ROS-Unity Handshake received" on the ROS side. Because the docker uses arbitrarily sets the "Override Unity IP Address" to 127.0.0.1, the ROS connection server on the Unity side will skip the handshake process, as shown in [the code](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/main/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L216). Therefore, the ROS side will not print the message "ROS-Unity Handshake received". We will update the troubleshooting part so that it could be clearer for readers.

Please let us know if it works fine on your machine.

peifeng-unity, 2021-03-02:
Hello @engineering-coder, it is good to hear your interest in using docker and trying the ROS connection. I was under the wrong impression of the handshake process, please ignore my previous answer if you have seen them.

It seems that all look correct in the configurations of ROS Connection and the printouts in the docker container. Can you double-check and make sure that the port 10000 and 5005 are exposed and accessible from your host to the container? e.g. the firewall rules.

Depending on your docker settings on Windows, you may need to use the NATed IP instead of the loopback IP address. We may able to diagnose more if you could share more information about your docker setups.

engineering-coder, 2021-03-02:
> Hello @engineering-coder, it is good to hear your interest in using docker and trying the ROS connection. I was under the wrong impression of the handshake process, please ignore my previous answer [...]

Thanks for replying! I am a new user of Docker and may not be able to provide enough information in this single reply.

I am using Docker Desktop (Engine Version 20.100.2) on Win10 Home 1909 (18363.1379).

I am using the command line in the tutorial "docker run -it --rm -p 10000:10000 -p 5005:5005 unity-robotics:pick-and-place /bin/bash" to start the Container and I believe that the ports 10000 and 5005 on the host and container are connected but I am not sure about the Windows firewall settings (and have no idea how and where to check the settings).

I have attached some Docker settings that may be helpful:
[image]
[image]
[image]
[image]

engineering-coder, 2021-03-02:
> Hello @engineering-coder, it is good to hear your interest in using docker and trying the ROS connection. I was under the wrong impression of the handshake process, please ignore my previous answer [...]

Besides, I am getting the "No data available on network stream after 10 attempts." Error on the Unity Console, just a few minutes after pressing play button.
[image]

peifeng-unity, 2021-03-02:
Hello @engineering-coder, thanks for all the rich information. The error message on the Unity side is from the handshake process when the Unity server attempts to pull the response from ROS and it consistently fails. We would recommend the following steps to check whether everything will work fine.

1. Try our [quick_demo tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/quick_demo.md) and see if it works
2. If the quick demo doesn't work, please try to downgrade Unity's version and try it again with Unity 2020.2.0b9
3. If it still doesn't work, it would probably be a communication issue. There are a couple of settings to try that we can think of at this moment:
    -  we noticed that your [WSL2 integration is not correctly configured](https://user-images.githubusercontent.com/34475872/109585623-ce0bca00-7b3e-11eb-8695-ea83e7527f64.png) because no WSL2 distro is installed. Configure it correctly may help to forward the ports
    - please double-check the [inbound rules of the Windows firewall](https://docs.microsoft.com/en-us/windows/security/threat-protection/windows-firewall/create-an-inbound-port-rule) and make sure the ports are open to docker

Please let us know how it goes?

engineering-coder, 2021-03-08:
> Hello @engineering-coder, thanks for all the rich information. The error message on the Unity side is from the handshake process when the Unity server attempts to pull the response from ROS and it [...]

Thanks a lot for your advice! But it still doesn't work on my machine.

I have downgraded my Unity to 2020.2.0b9, installed and integrated a WSL2 distro (ubuntu 18.04), and set an inbound rule for Windows firewall to make sure that the ports are open to docker. But the problem still exists.

I tried to run 'telnet 127.0.0.1 10000' on my host machine (running win10), and the results showed that the port is reachable from windows.

However, when I tried 'curl 127.0.0.1:5005' or 'curl 127.17.0.2:5005' in the contained, it returned: 'curl: (7) Failed to connect to 127.0.0.2 port 5005: Connection refused' or the similar. Then I found [this site](https://stackoverflow.com/questions/24319662/from-inside-of-a-docker-container-how-do-i-connect-to-the-localhost-of-the-mach) and tried to run 'curl docker.for.win.localhost:5005' and 'curl host.docker.internal:5005' in the container. It returned 'curl: (52) Empty reply from server' if Unity is not running in the Windows.

If unity is running before 'curl docker.for.win.localhost:5005' is run, no message will be shown in the console of the container and the console of Unity. Once I terminated the curl command in the container, the error message "No data available on network stream after 10 attempts." will show up in Unity. So I believe that the problem is the localhost of the container is only reachable via 'docker.for.win.localhost' instead of '172.17.0.2' inside the docker container.

However, when I tried 'echo "ROS_IP: docker.for.win.localhost" > src/niryo_moveit/config/params.yaml' and run the launch file afterwards, the node returned an error. Maybe it is an issue regarding the source code of the node.

Maybe this can provide some possible solutions to the problem? I will be very happy to know if this information helps!

[image]

peifeng-unity, 2021-03-12:
Hello @engineering-coder, sorry for the late response. It is great to hear about your progress in debugging. As far as I understand, you are believing that the IP address is not set correctly and it makes sense to us from the information.

For the error messages you have attached, you cannot use a string as the IP address in Unity (e.g. `docker.for.win.localhost`) because the `ros_tcp_endpoint` package only accepts the dotted-decimal format IP address, and it throws the errors that you've attached. The `docker.for.win.localhost` should be the DNS name of the docker gateway in the docker network. You can get its IPv4 address by simply pinging it (e.g. `ping docker.for.win.localhost`), but please don't use this IPv4 address in the ROS Settings.

It seems that the docker on your Windows machine cannot recognize the loopback IP address (e.g. in the NAT mode). Another option is to use your machine IP instead of the loopback IP. Here is [an instruction about how to find the machine IP address](https://support.microsoft.com/en-us/windows/find-your-ip-address-f21a9bbc-c582-55cd-35e0-73431160a1b9).

You can see that there are two IP addresses in the ROS Settings: the `ROS IP` and `Override Unity IP Address`. You can try to change both of them to your machine IP address. If it still doesn't work, please try to set `ROS IP` to your machine IP and leave the `Override Unity IP Address` blank.

Hope this could help and please let us know if it works?

engineering-coder, 2021-03-13:
> Hello @engineering-coder, sorry for the late response. It is great to hear about your progress in debugging. As far as I understand, you are believing that the IP address is not set correctly and it makes sense to us from the information.
> 
> For the error messages you have attached, you cannot use a string as the IP address in Unity (e.g. `docker.for.win.localhost`) because the `ros_tcp_endpoint` package only accepts the dotted-decimal format IP address, and it throws the errors that you've attached. The `docker.for.win.localhost` should be the DNS name of the docker gateway in the docker network. You can get its IPv4 address by simply pinging it (e.g. `ping docker.for.win.localhost`), but please don't use this IPv4 address in the ROS Settings.
> 
> It seems that the docker on your Windows machine cannot recognize the loopback IP address (e.g. in the NAT mode). Another option is to use your machine IP instead of the loopback IP. Here is [an instruction about how to find the machine IP address](https://support.microsoft.com/en-us/windows/find-your-ip-address-f21a9bbc-c582-55cd-35e0-73431160a1b9).
> 
> There are two IP addresses in the ROS Settings: the `ROS IP` and `Override Unity IP Address`. You can try to change both of them to your machine IP address. If it still doesn't work, please try to set `ROS IP` to your machine IP and leave the `Override Unity IP Address` blank.
> 
> Hope this could help and please let us know if it works?

The problem is solved after I set 'ROS IP' and 'Override Unity IP Address' into my machine IP! Thanks a lot for your help!

engineering-coder, 2021-03-15:
> Hello @engineering-coder, sorry for the late response. It is great to hear about your progress in debugging. As far as I understand, you are believing that the IP address is not set correctly and it [...]

Sorry to bother you but I think I have closed this issue too quickly. This solution worked on Win 10 Home 1909, as I stated in the previous comment, but when I upgraded to 20H1 and later to 20H2, the solution no longer works.

When I enter play mode in Unity, 'ROS-Unity server listening...' printed on the Unity side but no 'ROS-Unity Handshake received' on the ROS side. and later I will receive a 'No data available on network stream after 10 attempts.' error in Unity Console.

And when I press the "Publish" button, another error pops up in Unity Console saying 'TCPConnector Exception: System.IO.IOException: Unable to write data to the transport connection: An existing connection was forcibly closed by the remote host.' This never happened before in Win10 Home 1909 (even when the connection is not successful and no handshake is received by ROS), and I have no idea what is going wrong.

I have set up an inbound rule and an outbound rule to allow Docker and Unity to connect, and I have turned off the Windows Firewall.

I hope to know if you know anything about this problem! 

The pictures contain Chinese characters because my Windows licence only allows Chinese as the displaying language. "IPv4 地址" means "IPv4 Address" and "远程主机强迫关闭了一个现有的连接。" means "An existing connection was forcibly closed by the remote host.".

[image]
[image]

peifeng-unity, 2021-03-15:
Hello @engineering-coder, thanks for reporting this issue. This still looks like causing by IP addresses or firewall rules, and it is difficult to tell without further information. You could try the following:

- make sure that all the firewall and antivirus software allows communication (IP, ports, protocols, etc.) between Unity and the docker container; if you are using your own docker image, please also make sure to turn off the Ubuntu firewall as well
- Make sure that ROS servers are listening to the correct IP and port
- You could retry to use the loopback IP and check if it works on the new version of Windows

Please let us know if any of the above may help. Thank you.

peifeng-unity, 2021-04-06:
Hello @engineering-coder, we just wish to follow up and double-check whether the issue has been solved? Thank you for letting us know.

engineering-coder, 2021-04-07:
> Hello @engineering-coder, we just wish to follow up and double-check whether the issue has been solved? Thank you for letting us know.

Thank you and sorry for the late update. I have turned off the firewall, set the inbound and outbound rules, and tried both the loopback IP and machine IP. But the problem still exists.

As for our project, I have transformed my testing and debug process entirely to the Ubuntu edition of Unity, and the connection between two machines (one running Ubuntu and another running Windows 10) worked fine. So this problem no longer bothers me.

If it matters to you, I can close this issue.

peifeng-unity, 2021-04-08:
Thank you @engineering-coder for letting us know that you have a way to work around the issue. I was asking the question just because we are going to provide the [troubleshooting guidance for ROS connections](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/71fedec52a6226a2bcd9c3a1654832f5c18b96e8/tutorials/ros_unity_integration/network_troubleshooting.md) and wish to let you know if you haven't made it work yet.

peifeng-unity, 2021-04-16:
Closing because it has been inactive for a while. Please feel free to reopen it if there are more questions.

closed 2021-04-16 by peifeng-unity

DGAD7, 2023-02-26:
Hey guys, I have managed to solve this problem. But not sure if the root cause is generic.
When this problem occurs, I guess accessing localhost from a web browse returns an error something like "Connection refused". localhost is 127.0.0.1 which most of us use to connect to docker.
Seemed like something was blocking my request. After some googling around , figured a way out to know which windows processes (PID) is also using the port 10000.
Installed XAMPP tool and ran netstat option and figured out that svchost.exe was also using the port 10000. 
netstat also shows the pid of the process, in my case it was 5304.
opened cmd promot as admin and ran this command to kill that pid.
taskkill /pid 5304 /f

And now unity connects to ros on docker.

Had to do this again after a restart but not sure if this has to be done everytime. 

fyi, it was all working fine for me unitil 3 days ago when I could not longer connect to ros from unity.
So windows update i would guess. Another reason why Linux/Mac are favourites!

DGAD7, 2023-02-26:
an update, svchost.exe grabs the port 10000 everytime unity releases it. i.e everytime game mode is exited and started again.
Not sure what is causing this suddenly and how to permanently solve this. but the temporary solution works for now.
[image]