Unity-Technologies/Unity-Robotics-Hub #171: Tutorial for the Pick and Place does not compile

darksnakezero, 2021-03-01:
Expected Behaviour:

Following the tuorial, the Pick and Place should be imported correctly.

Actual Behaviour : 
On importing the PickandPlaceProject folder, an error prevents the code from being executed.

Steps to reproduce:

Set up acompletelynew Unity installation on Windows 10.
Try to follow the PickandPlace tutorial. 
On the step to import the project, it will fail.

Workaround Part 1:
Import the Editor Coroutines in the UnityPackageManager.
I would have expected that to be mentioned in the tutorial.

-> Still, Compilation fails. Now at line 188 at the Demo.cs file.

Workaround Part 2: Replace 2 Lines in Demo.cs with one
old lines:
188 var robotImporter = UrdfRobotExtensions.Create(<THREE arguments, cant be bothered to go back to that state right now, sry>);
189 while (robotImporter.MoveNext()) {}

replace with 

188 UrdfRobotExtensions.Create(urdfFilepath, urdfImportSettings);

Now it compiles without errors and actually works.

LaurieCheers-unity, 2021-03-01:
Hi, you've somehow managed to check out an older version of the URDF-Importer package. On line 8 of the file PickAndPlaceProject/Packages/manifest.json, it should say:

    "com.unity.robotics.urdf-importer": "https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.2.0",

What does yours say?

darksnakezero, 2021-03-02:
Hi,

that is true, It seems, I did check out version 0.1.2. 

I checked out the repository again and began anew. 

It now imports 0.2.0 correctly and runs smoothly.

Unfortunately, I did not manage to reproduce how I managed to get version 0.1.2 despite using only these tutorials.

It works now, thank you very much

closed 2021-03-02 by darksnakezero