Unity-Technologies/Unity-Robotics-Hub #173: MoveIT Solver is not aware of robot unity position?

mrceresa, 2021-03-02:
Dear all,
thanks for the great examples and apologies if I ask a silly question. 

If I move the robot arm in unity, how can I make the solver aware that its position changed? It happily continues to plan a trajectory even if the targets are now out of reach
[image]

Thanks and regards,

Mario

LaurieCheers-unity, 2021-03-02:
Hi there; yes, this demo makes the assumption that the robot has not been moved in Unity.

If you'd like the robot to be at a different fixed position you can tell MoveIt about it by adjusting its position in the URDF file (https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/ROS/src/niryo_one_urdf/niryo_one.urdf.xacro#L171).

If you need a dynamic solution, try subtracting the robot's position from the positions reported in TrajectoryPlanner. PublishJoints.

mpinol, 2021-03-15:
Looks like the question was answered. Please don't hesitate to reach out if you have any other questions!

closed 2021-03-15 by mpinol