Unity-Technologies/Unity-Robotics-Hub #174: Key Error on server_endpoint.py

NathanielRose, 2021-03-03:
I am getting a KeyError for `/ROS_IP` on my demo server that was generated in catkin. No changes were done to ros environment or catkin environment.

```
Traceback (most recent call last):
  File "/mnt/c/Users/rosen/OneDrive/Desktop/Imperial/MRes_Research/Projects/catkin_space/src/robotics_demo/scripts/server_endpoint.py", line 26, in <module>
    main()
  File "/mnt/c/Users/rosen/OneDrive/Desktop/Imperial/MRes_Research/Projects/catkin_space/src/robotics_demo/scripts/server_endpoint.py", line 13, in main
    tcp_server = TcpServer(ros_node_name, buffer_size, connections)
  File "/mnt/c/Users/rosen/OneDrive/Desktop/Imperial/MRes_Research/Projects/catkin_space/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/server.py", line 45, in __init__
    self.tcp_ip = rospy.get_param("/ROS_IP")
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 467, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: '/ROS_IP'
```

Any tips on how to repair?

LaurieCheers-unity, 2021-03-03:
Hi there, the ROS-TCP-Endpoint server expects a rosparam called ROS_IP to be defined. You could do this by simply running `rosparam set ROS_IP YOUR.IP.ADD.RESS`, for example.

In our Pick and Place demo there's a nicer setup where the parameter initialization is done automatically by the .launch files. For example: https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/ROS/src/niryo_moveit/launch/part_3.launch

Here the command `<rosparam file="$(find niryo_moveit)/config/params.yaml" command="load"/>` loads the parameters defined in this file:
https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/ROS/src/niryo_moveit/config/params.yaml

NathanielRose, 2021-03-10:
@LaurieCheers-unity Thanks, the `rosparam set` turned out to fix it. For some reason upon launching, my tcp server endpoint was not setting this.

closed 2021-03-10 by NathanielRose