Unity-Technologies/Unity-Robotics-Hub #178: Is there a way to ensure a grasped object doesn't slip away?

karthikm-0, 2021-03-09:
Hi,

I have a question regarding grasping a rigidbody. I've found that for my specific robot, the grasping is quite inconsistent. Most of the time, the robot attempts to pick up the object but it will slip out of its reach. Other times, it is able to hold on to the object. I was wondering if anybody has suggestions about how I might go about ensuring a successful grasp all the time, particularly the case where the object slips after it has already been held. Which parameters could be tuned to ensure its success? Thanks!

mrpropellers, 2021-03-10:
Hi @karthikm-0, while there's no 100% effective fix that I know of, there should be some parameters for surface friction on your grippers that can be increased to make them "stickier." I'll consult one of our PhysX integration experts as well, to see if they have any additional advice...

karthikm-0, 2021-03-10:
Thanks for the response, @mrpropellers. The reason behind my question is I'm building an interface to support the teleoperation of robot arms so the emphasis is on the interaction aspect, under the assumption that the object stays grasped if it was initially picked up.

mrpropellers, 2021-03-11:
If your experiment assumes that it is not possible for the grasper to drop the object while in transit, the only way to ensure this 100% of the time would be to script it. I'd recommend adding some logic to a monobehaviour that attaches your cube to the grasper's transform tree and disables physics on the cube. Then, when the grasper releases the cube, reset its transform parent to the global space and re-enable physics.

karthikm-0, 2021-03-12:
Thanks for the response, @mrpropellers. I tried this approach, where the object is made kinematic upon grasping, but I found that the gripper continued to apply force on to the kinematic object until it fell down again.

mrpropellers, 2021-03-13:
Hmmm you may need to also disable collisions on the object's collider. Likely the easiest way to do this is to fetch a reference to the MeshCollider and set `IsTrigger` to True.

peifeng-unity, 2021-04-08:
Hello @karthikm-0, if you do not care whether the simulation is physically realistic and just want 100% sure that the object won't slip out, the workaround could be making the object an additional joint of the robot when it has been grasped. In this case, the object is guaranteed not to slip. Please let us know if this workaround may help.

mrpropellers, 2021-04-13:
Closing due to lack of activity.  @karthikm-0 hopefully you've reached a solution that enabled your use case.  Feel free to re-open if you need more troubleshooting help.

closed 2021-04-13 by mrpropellers