Unity-Technologies/Unity-Robotics-Hub #179: ToString overload Error after importing ROS TCP Connector package in Unity

Jordi24, 2021-03-10:
Hi, after following the steps as specified in [ROS–Unity Initial Setup](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#unity-scene), I get the following errors/warnings when importing the ROS TCP Connector package:
```
error CS1501: No overload for method 'ToString' takes 2 arguments
```
[image]

In the specified files, I believe it concerns these lines of code:
```
public string ToString(string format) => internalVector.ToString(format);
public override string ToString() => internalVector.ToString();
public string ToString(string format, System.IFormatProvider formatProvider) => internalVector.ToString(format, formatProvider);
```

Do you have an idea what causes this issue and how it could be solved?`

Possible relevant info: using Unity 2019.3.6f1

mrpropellers, 2021-03-10:
Hi @Jordi24, it's difficult to know for sure as  we haven't tested the Unity version you're running but it seems likely there is interface mismatch here due to the older Unity version.  It looks like we're not calling this out explicitly on that setup page, but our recommended version is 2020.2+ -- you may get away with some 2020.1 versions, but earlier versions are unlikely to work without some hacking on the code, and many of the physics integrations we rely on for stable simulations were not yet introduced.

mrpropellers, 2021-03-12:
Hi @Jordi24 , did opening the project with a more recent version of Unity resolve this issue?

Jordi24, 2021-03-12:
Hi @mrpropellers, I wanted to try your suggestion to use a more recent version during the weekend. I will inform you as soon as I have some results.

Jordi24, 2021-03-14:
@mrpropellers Upgrading to Unity Version 2020.3.0f1 LTS solved the issue. Thanks for the help!

closed 2021-03-14 by Jordi24