Unity-Technologies/Unity-Robotics-Hub #181: prediction move robot

fahrulmr, 2021-03-11:
hi, 

im working on digital twin arm robot
i found this link 
https://new.siemens.com/global/en/company/stories/research-technologies/digitaltwin/robotics-simulation.html

thats what i need
https://assets.new.siemens.com/siemens/assets/api/uuid:1f8e967d-ddb6-4705-8ad6-3f9654e8a21d/width:1266/quality:high/martin.gif

are you use base pick and place demo?

how to make prediction like .gif,that awesome 
can you give me clue about this
so real robot move, and in unity thats will prediction motion and very acurat movement

thanks
fahrul

mrpropellers, 2021-03-11:
Hi @fahrulmr, while I can't speak to the specifics of Siemen's implementation, what they are demonstrating would be possible via extension of the pick and place example we have in this hub.  We are currently prototyping message visualizers that are somewhat similar to Siemen's example, [but these are still in a side branch of the TCP connector package](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/laurie/PoseVisualizers). We also recently added a [Part 4](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/4_pick_and_place.md) to the Pick and Place tutorial, which demonstrates how we extended the example to send the same trajectory commands to the real arm in the same way they were sent to the simulation.

mrpropellers, 2021-03-12:
Closing this issue as there's no action to take. @fahrulmr keep an eye on the Unity blog for updates - we plan to open a Robotics section of the Unity forums soon which will better facilitate discussions like this.

closed 2021-03-12 by mrpropellers