Unity-Technologies/Unity-Robotics-Hub #187: URDF imported objects lose mesh colliders when made into prefabs

gbattra, 2021-03-23:
**Unity Version: ** 2020.3.0f1

If I make the imported robot arm into a prefab, it loses its mesh colliders.

Steps to reproduce:

1) Open up the `pick-n-place` tutorial project
2) Import the Niryo One arm using the URDF importer
3) Drag the new `niryo_one` instance into the `prefabs` directory
4) Open the newly made prefab and check the `mesh_collider` of any link (the mesh reference will be `None`)

Example:
[image]

vidurvij-Unity, 2021-03-23:
Hi @gbattra,
Thanks for bringing this to our attention. I was able to reproduce this error. It seems that when the Robot is imported using VHACD decomposer, the meshes are saved in Assets because of which the prefab loose reference to the mesh files when reloaded. 
We will have fix for this issue as soon as possible.
Vidur

vidurvij-Unity, 2021-04-13:
@gbattra the issue has been fixed, Its in our dev branch and we update you as soon as it is merged in the main.

vidurvij-Unity, 2021-05-10:
@gbattra You should use version v0.3.0 of the URDF Importer. It has the fix in it. If the problem persists feel free to reopen the issue.

closed 2021-05-10 by vidurvij-Unity