Unity-Technologies/Unity-Robotics-Hub #190: Can the DockerFile be shared

NathanielRose, 2021-04-06:
In the tutorials theres a docker image that is used as an alternative for the pickandplace example.
Can the DockerFile used for the image: `unity-robotics:pick-and-place` be shared. Attempting to get this to work on WSL has become quite a hassle.

peifeng-unity, 2021-04-06:
Hello @NathanielRose, the DockerFile is located in tutorials/pick_and_place/docker/Dockerfile of the repository and you can use [this link](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/docker/Dockerfile) to find it. Please let us know if it helps.

NathanielRose, 2021-04-07:
Thank you, for some reason this wasnt showing in the search results.

closed 2021-04-07 by NathanielRose