Unity-Technologies/Unity-Robotics-Hub #192: Plans to include XML-RPC similar to Iviz?

tgroechel, 2021-04-06:
I stumbled across [Iviz](https://github.com/KIT-ISAS/iviz/tree/devel) recently after moving off of [ROS#](https://github.com/siemens/ros-sharp). Are there plans to incorporate a similar protocol? The nicest part of Iviz is that you don't need to run anything extra on the ROS side of things. I haven't benchmarked any latency differences yet but will likely in the near future.

hyounesy, 2021-04-07:
Hi @tgroechel , Thanks for bringing this to our attention. 
No, we currently don't have an immediate plan to have a native XMLRPC support. But we will investigate this if we find that the benefits outweigh the maintenance costs. In the meantime please keep up posted on your findings.
What are your main reasons for looking for ros-sharp alternatives?

tgroechel, 2021-04-13:
Hi @hyounesy, that sounds great and understand not wanting to add maintenance cost. I do [Virtual, Augmented, and Mixed Reality for Human-Robot Interaction (VAM-HRI)](https://vam-hri.github.io/) research and we are exploring the different communication protocols/libraries (this, ROS#, and Iviz/XMLRPC). Our group in particular is mostly looking at ease of maintenance/ramp up time for new members, support for older versions of ROS (namely indigo), and latency for messages (namely image/depth).

A-Ivan, 2021-05-05:
Hi @tgroechel 

This is a topic I am also interested in this topic (VAM-HRI). Have you looked at the ros_reality repository? They have worked on both AR and VR applications with Unity and ROS, their implementation is a bit older (2019) so they used the ROS# library, but I think it could easily be ported over to Unity's current library (this repo).

Here is the ros_reality main repo page - https://github.com/h2r/ros_reality
Here is their use with the Hololens - https://github.com/dwhit/ros-sharp

I didn't know about iviz, that looks really interesting.

hyounesy, 2021-07-08:
Closing this issue as there is no further action to take at this moment. Feel free to open a new issue if you had any updates or questions.

closed 2021-07-08 by hyounesy