Unity-Technologies/Unity-Robotics-Hub #193: Tutorial or example of UnityService (set Unity as the server in service)

Mingrui-Yu, 2021-04-07:
There are nice tutorials about how to set publiser, subscriber in ros-unity integration in [ros_unity_integration/README.md](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/README.md). As for the service, I can only find an example which sets the ROS as the server and Unity as the client. 

Is there any example about how to set the ROS as the client and Unity as the server? What is the code to achieve this in Unity scripts?

Thank you very much!

hyounesy, 2021-04-07:
Hi @Mingrui-Yu, 
Could you please provide more information on this?
The ROS-Unity integration assume that a ROS server is running outside unity environment, and provide the means to subscribe or publish to ros topics using [ROS-TCP-Connector](https://github.com/Unity-Technologies/ROS-TCP-Connector) unity package and [ROS-TCP-Endpoint](https://github.com/Unity-Technologies/ROS-TCP-Endpoint) ros node.
So, how do you imagine using the Unity as the server? Do you have a specific use case in mind?

Mingrui-Yu, 2021-04-08:
Thanks for your reply. Sorry that my question is not so clear. I just mean the communication type "service" in ROS, in which  Unity component receives a Request message from another ROS node and then sends a Response back. 

As in [server_endpoint.md](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/server_endpoint.md), there are "ROS Publisher", " ROS Subscriber", "ROS Service" and "Unity Service", but only the tutorials/examples of the first three are provided in [ros_unity_integration/unity_scripts](https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/main/tutorials/ros_unity_integration/unity_scripts).  I'd like to know what is the code to achieve the above "service" communication where Unity component receives a request and then sends a response back. Thanks!

hyounesy, 2021-04-12:
Thanks for the clarification @Mingrui-Yu . I am looking at this now. Will provide you with a example code soon.

hyounesy, 2021-04-15:
Hello @Mingrui-Yu . An example for [creating a UnityService](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/ros_unity_integration/unity_service.md) has been added to the [ros_unity_integration tutorials](https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/dev/tutorials/ros_unity_integration)  (currently on the #dev branch).
Inorder to use this you need to use the #dev branch of the [ROS-TCP-Endpoint](https://github.com/Unity-Technologies/ROS-TCP-Endpoint/tree/dev) on the ROS side and also install the dev branch of [ROS-TCP-Connector](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/dev) Unity package.
Please let us know if you had any further questions.