Unity-Technologies/Unity-Robotics-Hub #196: RosSubscribe Timing out

Koriath, 2021-04-08:
I'm trying to create a subscriber for /joint_states using an UR3 robot, with ROS running on a linux computer and Unity running on a separate Windows computer. I believe my setup is currently working because I can call services (like the one to play/stop the robot) with no issues.
However, when I added a subscriber to my default_endpoint_server.py, unity did not receive any data.
The unity script I have:

`void Start()
    {
        ros = ROSConnection.instance;
		ros.Subscribe<JointState>("joint_st", SubscribeJointStates);
    }`

`void SubscribeJointStates(JointState js)
	{
		Debug.Log("Joint State Received:" + js);
		if (js.position.Length > 0)
		{
			position.x = (float)js.position[0];
		}
	}`

The "position.x" is just a Vector3 I'm using to debug this.

on my default_server_endpoint.py I have added this line:

`'joint_st' : RosSubscriber('joint_states',JointState,tcp_server)`

If I simply start everything, ROS works fine with no errors, Unity launches and connects sucesfully but, the TCP-Endpoint will constantly printout errors:

`[INFO] [123876443.49875]: Exception timed out`

So I thought that maybe there were firewall issues, or something going on with the connection, but this is not the case.
I fired up Hercules and set him to listen on port 5005. Not only does Hercules receives the JointStates correctly, the time out errors stop! As soon as turn off Hercules and start my unity application however, the TCP-Endpoint will printout a message about the Unity Handshake being correctly established and... resume timing out.

I don't necessarily need to subscribe to /joint_states for my application, but it would help me further down the line.
Is there anything I've done wrong? Some setup I've missed?

Koriath, 2021-04-08:
Okay, something really weird must be going on.
I tried to subscribe to something even simpler: the Robot Mode topic.
It should only consist of a single variable with a number for the state.
Unity behaves just like before (not receiving anything) and the TCP-Endpoint still outputs the timed out error.
But when I switch back to using Hercules on port 5005, well:

[image]

I'm guessing that THIS is not the data I'm suposed to be receiving. It's just returning the type of msg, instead of the message itself. What am I doing wrong?

Koriath, 2021-04-09:
Out of curiosity, I attempted to invert my code and turn that Robot Mode subscriber into a publisher.
It doesn't really make sense for me to publish the Mode of the Robot, but it's a quick way to check if publishing is working.
I successfully managed to publish it (and funny enough, the UR Driver I'm using didn't like values above 7 and promptly crashed, but that was to be expected).

So, as of now, I can publish data and call services. But subscribing is still a no-go.
If anyone here can help, I'd appreciate it very much. In the meantime, I'll try to create a service that gives me the /joint_states as a response for a temporary quick hack.

Koriath, 2021-04-09:
Alright, I found the cause. It's my fault, but I'd like to recommend a change to the tutorials/examples.
In the "robotics_demo", the server_endpoint.py" has a like like this:

''color': RosSubscriber('color', UnityColor, tcp_server)'

And on the unity side, for the Subscriber tutorial, this Subscriber is acessed with:

'ROSConnection.instance.Subscribe<RosColor>("color", ColorChange);'

The first impression I had with this function, is that when Unity specifies "color" as the topic, it's doing so because we defined a Subscriber named "color" in the server_endpoint.py.
I based this on the fact that when I call a service, I need to use the service name I assigned to it in the endpoint, NOT the actual service name in the ROS hierarchy.

After a few desperate hours, I tried using:

'ros.Subscribe<RobotMode>("ur_hardware_interface/robot_mode", ReadRobotState);'

The service is provided by the UR driver.
This has worked flawlessly, and the callback function is executed. 
I'd recommend changing the tutorial to less ambiguous terms like these, or at least include a note somewhere about it.

Another note about the same topic: ROS does let you subscribe to topics that haven't been published yet (or that are simply empty). But when doing so, ROS typically gives me a warning about it, "This topic might not have been published yet".
If this was logged on the unity side too, I probably would have caught early on that I was not subscribing to the right topic.
Please consider adding it in the future. Thank you for the rest of the code though, asides from this debacle, it's been a real pleasure working with it so far.

hyounesy, 2021-04-09:
Hi @Koriath. Thanks for bringing this to our attention and thanks for the great suggestions. I have filed a couple internal tickets to address the clarity in the document and logging on the unity side.

As for your connection issue, I suspect it might be related to IP/port or firewall settings. What are the settings in your Unity ROS Settings (more specifically "ROS IP Address", "ROS Port", "Overrride Unity IP Address", "Unity Port") and the rosparam values for `ROS_IP` and `ROS_TCP_PORT`?

Koriath, 2021-04-15:
In my initial setup:
ROS IP Address : 192.168.10.137
ROS Port: 10000
Override IP: 192.168.10.127
Unity Port: 5005

ROS_IP: 192.168.10.127
ROS_TCP_PORT: 10000

However, I still had the same problem. Kept timing out. So I ended up switching everything to the Linux side (works well, no time out) and I'll control it from windows with a couple TCP sockets.
I tried to find a causa but was not successful. It might be similar to the problems some folks are having when working with ROS and WSL (they basically emulate Linux on a Windows machine and apparently there's a heap of problems doing so).

One additional note: I noticed that the message generation can import "actions" but doing so causes errors (double definitions and things that are already imported by default ?) but there's no documentation about using Actions on this Repo.
Are they not supported?

vidurvij-Unity, 2021-05-10:
@Koriath For ROS actions you may want to lookup this issue. https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/202

hyounesy, 2021-07-08:
I am closing this issue. However If your problem continues , feel free to reopen or create a new issue.

closed 2021-07-08 by hyounesy