Unity-Technologies/Unity-Robotics-Hub #200: Ros Time in Unity

turtleMaster20, 2021-04-15:
Hello,

I am publishing some topics with headers in Unity. How do I assign a time to this header? Can I also sync this time with the time in ROS?

Thanks in advance.

peifeng-unity, 2021-04-16:
Hello @turtleMaster20, thanks for bringing this to our attention. I have added a ticket and we will reply to you soon.

LaurieCheers-unity, 2021-04-19:
Hi, we don't currently have built-in support for the ROS clock but it's definitely on the roadmap. The following code should help you convert between a C# DateTime and a ROS Time message:

        DateTime ToDateTime(Time message)
        {
            DateTime time = new DateTime(message.secs);
            time = time.AddMilliseconds(message.nsecs / 1E6);
            return time;
        }

        Time ToROSTime(DateTime dateTime)
        {
            return new Time { secs = (uint)dateTime.Ticks, nsecs = (uint)(dateTime.Millisecond * 1E6) };
        }

You can then set your header's time field directly, or pass a time into its constructor like this: new Header(seq, ToROSTime(DateTime.Now), frame_id)

To sync with the ROS time, I suggest you subscribe to the ros "/clock" topic: whenever a new clock message is received, remember how far ROS time was from Unity time, and then use that offset to correct the time in the messages you create.

turtleMaster20, 2021-04-20:
Thanks a lot, I will close the issue now

closed 2021-04-20 by turtleMaster20