Unity-Technologies/Unity-Robotics-Hub #201: ROS Connection Problem

Ariansdf, 2021-04-15:
Hi,
I followed the pick and place tutorial and I believe everything is installed correctly. 
As I launch "niryo_moveit part_2.launch", I get:
```
process[rosout-1]: started with pid [8547]
process[server_endpoint-2]: started with pid [8550]
process[trajectory_subscriber-3]: started with pid [8551]
[INFO] [1618483306.209545]: Starting server on 10.170.50.22 2001:638:611:2170:b095:9fb8:9c66:2019 2001:638:611:2170:2577:3239:6744:25ed:10000
// and it stays at:
socket.gaierror: [Errno -2] Name or service not known
```
On the Unity side I get the error:
`SocketException: Connection refused`

My firewall is also deactivated and I've double checked IP and Port properties.
Can you please explain what's going wrong?

[image]

Ariansdf, 2021-04-15:
I have also manually changed params.yaml to ROS_IP: 192.168.50.149. In this case i get this error on ROS side:
`OSError: [Errno 99] Cannot assign requested address`

and nothing in Unity Console.

[image]

A-Ivan, 2021-04-16:
Try changing the ROS IP Address to something else. Set your IP using a static IP to a different value, then set that value in both the params file and in Unity.

peifeng-unity, 2021-04-16:
Hello @Ariansdf, thank you for bringing this to our attention. It looks like the tcp_server on the ROS side cannot bind to the given IP address and port. I have added a ticket to our team and will reply to you soon.

peifeng-unity, 2021-07-13:
Hello @Ariansdf, sorry for the so late response. I don't know if you were able to resolve the issue, but our team has updated and improved the ROS connection method so that there will be fewer glitches to set it up. Please feel free to try [the dev branch of the pick-and-place tutorial ](https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/dev)and let us know if the issue is still there.

Another possible solution is to set the ROS_IP parameter to 0.0.0.0 since it seems that you are using Ubuntu.

sdiao, 2021-08-02:
We have made several improvement since the issue was open. Please checkout the latest release 0.5.0 and try the tutorial again. Close this issue due to inactivity.

closed 2021-08-02 by sdiao

bennyNMIS, 2024-06-17:
hi, what was the solution to this problem? im having the same trouble @Ariansdf @peifeng-unity @sdiao