Unity-Technologies/Unity-Robotics-Hub #202: Support for Ros actions

omarbenjem, 2021-04-15:
Hello,

I am trying to implement a Ros action to move an object in Unity. There is already the support for Ros services but I am looking for a preemptible service, basically a Ros action. Is a support (and tutorial) coming soon for Ros actions? Is there currently another way to implement this?

Thank you very much!

peifeng-unity, 2021-04-16:
Hello @omarbenjem, thanks for bringing this up to our attention. You should be able to generate the Action messages but will have to publish or subscribe to the action's topic. [This part of the tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/4_pick_and_place.md) uses actions and it might be a good reference. Please feel free to let us know if you have more questions.

vidurvij-Unity, 2021-05-10:
I hope the solution by @peifeng-unity helped you out. I am closing this issue, If you face further problems , feel free to open a new issue.
Vidur