Unity-Technologies/Unity-Robotics-Hub #203: Part4 : 'ROSGeometry' and 'ROSConnection' could not be found

TurtleIndigo, 2021-04-15:
Hello everyone,

I managed to complete the 3 first parts but I am locked at the 4th.

At the Unity Step, I built ROS messages, I copied and pasted the Script_Part4 file in Assets and obtain theses errors in Unity: 
```
Assets\Scripts_Part4\RealSimPickAndPlace.cs(3,7): error CS0246: The type or namespace name 'ROSGeometry' could not be found (are you missing a using directive or an assembly reference?)
Assets\Scripts_Part4\RealSimPickAndPlace.cs(18,13): error CS0246: The type or namespace name 'ROSConnection' could not be found (are you missing a using directive or an assembly reference?)
```

Do you have any idea what went wrong ?

Thank you for reading

peifeng-unity, 2021-04-16:
Hello @TurtleIndigo, thank you for bringing this up to our attention. I have created a ticket for the team and we will try to reproduce the issue and reply to you soon.

LaurieCheers-unity, 2021-04-21:
Hi @TurtleIndigo, are you still using the same pick and place project setup from the previous stages of the tutorial? Which version of ROS TCP Connector do you see in the package manager? The errors you're getting suggest your installation is outdated or incomplete.

("The script don't inherit a native class" error is just caused by the compile errors, hopefully if we can fix those everything should be fine.)

TurtleIndigo, 2021-04-21:
Hi @LaurieCheers-unity, thank you for your answer. I think that I am using the very lastest versions of files and packages, concerning TCP Connector, I have the v0.2.0. 

However, I just managed to "fix" these errors by replacing, in _RealSimPickAndPlace.cs_ : `using ROSGeometry` by `using Unity.Robotics.ROSTCPConnector.ROSGeometry`. Also, I added `using Unity.Robotics.ROSTCPConnector` for the second error. I think it can work fine now. At least, errors disapeared and I can add this script to my game object.

What do you think ?

LaurieCheers-unity, 2021-04-21:
Ah, I see, good fix. :-)
This has been fixed in dev branch for months, but the fix will get merged to main today.

TurtleIndigo, 2021-04-23:
Ok great, thank you for your help

closed 2021-04-23 by TurtleIndigo

WalterRainey, 2021-04-23:
I sued the world and won. Then a racist CIA agent took my digital
government program and never paid me. I need help getting paid.