Unity-Technologies/Unity-Robotics-Hub #205: ROSConnection.SendServiceMessage can return async response value.

dasannikov, 2021-04-16:
We can call `ROSConnection.SendServiceMessage` as async and it can return response without using of callback. Like this:

``` cs
var request = new AddTwoIntsRequest(0, 22);
var response = await ROSConnection.instance.SendServiceMessage<AddTwoIntsResponse>("add_two_ints", request);

```
All you have to do is change function `SendServiceMessage`  to 
```cs
public async Task<RESPONSE> SendServiceMessage<RESPONSE>(string rosServiceName, Message serviceRequest,
            Action<RESPONSE> callback = null) where RESPONSE : Message, new()

```
and return `Task<RESPONCE>` instead `void` at the end of function

```cs
return serviceResponse;
```

peifeng-unity, 2021-04-16:
Hello @dasannikov, thank you for bringing up this to our attention and it is a good idea to return an async response instead of using the callback. I have created a ticket for the team and we will follow up on reviewing it.

In the meantime, you are very welcome to contribute to the repository following the [contribution guideline](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/CONTRIBUTING.md). Thanks again.

sdiao, 2021-06-01:
Hi @dasannikov, thank you for the suggestion. We have released the update to [`ROSConnection.SendServiceMessage`](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/v0.4.0/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L141-L160) in v0.4.0. Please check out the latest release v0.4.0 of [ROS-TCP-Connector](https://github.com/Unity-Technologies/ROS-TCP-Connector) and let us know if you have additional feedback. Thank you.

closed 2021-06-01 by sdiao