Unity-Technologies/Unity-Robotics-Hub #206: Part4 : Setting up Niryo One

TurtleIndigo, 2021-04-20:
Hello everyone,

I am using a Windows 10 PC with Unity and a Linux Noetic PC with ROS, I managed to do parts 1, 2 and 3 with this setup.
I am currently trying to do part 4 with the same setup but I have trouble to understand what to do.

The 2 steps in _Setting Up Niryo One - Add niryo_moveit Package_ are not clear for me :

> Update the niryo_moveit/config/params.yml file's ROS_IP parameter to match that of the Niryo One.
Do I have to change the IP parameter of my Linux PC file, my Niryo One SD file, my Windows PC file or all of them ?
What IP should I write : Linux PC IP, Windows PC IP or Niryo One IP (that I found in Niryo One Studio) ?

> Copy the niryo_moveit package to the catkin_ws directory on the Niryo One's catkin workspace at /home/niryo/catkin_ws and run the catkin_make command.
I unplugged the SD of the Niryo and chose to copy and paste the file from my Linux PC.
Do I need to paste it here : _/home/niryo/catkin_ws_ ou rather here : _/home/niryo/catkin_ws/src_ (where other files are situated) ?

Thank you for reading.

mpinol, 2021-04-21:
Hey @TurtleIndigo ,

You are correct, you do not need to use your Linux machine for this part. The Niryo One runs its own ROS stack on start up. 

Communication should go from Unity running on your Windows machine to the Niryo One so when updating the `niryo_moveit/config/params.yml` file you will want to put the IP of the Niryo One, the one found in Niryo One Studio. And yes you need to copy the `niryo_moveit` package, including the params file just updated, to the `/home/niryo/catkin_ws/src` directory on the Niryo One SD card. It looks like I may have missed an instruction, you will also need to copy the `ROS_TCP_Endpoint` directory to `/home/niryo/catkin_ws/src` as well.

And don't forget to update the `ROS_IP` set in Unity to point to the Niryo One.

I hope this helps and let me know how it works out!

TurtleIndigo, 2021-04-23:
Hi @mpinol,

Thank you for your answer, it is more clear for me.

I followed your advises and I get this error :
`TCPConnector Exception : System.Net.Sockets.ScoketException (0x80004005): No connection could be established because the target computer expressly refused it.`

I tried with hotspot connection and ethernet connection (changing IPs in Niryo and Unity each time) and I still get the same error.

mpinol, 2021-04-23:
Hey @TurtleIndigo 

Would you be able to connect the Niryo One to a wireless network and try again? You can find instructions on how to do this in the user manual, https://niryo.com/docs/niryo-one/user-manual/.

In the mean time I am going to see if I can reproduce your error with a hotspot connection.

TurtleIndigo, 2021-05-03:
Hello everyone,

For your information, I tried to redownload the entire folder and reset the scene with the new update v0.3.0.

I still get theses 2 errors on Play (the first is certainly due to the missing connection) :
> NullReferenceException: Objects reference not set to an instance of an object
(RealSimPickAndPlace.Start() (at Assets/Scripts_Part4/RealSimPickAndPlace.cs:149)

> SocketException:  No connection could be established because the target computer expressly refused it.

And still get this error when I click on the Publish button on Play (certainly due to the missing connection too)
> NullReferenceException: Objects reference not set to an instance of an object
(RealSimPickAndPlace.PublishJoints() (at Assets/Scripts_Part4/RealSimPickAndPlace.cs:108)

I can add a few screeshots, just to be sure I did not made a mistake somewhere.

IP adress on Niryo Studio: 
[image]

[image]

ROS Settings parameters in Unity:
[image]

File in src folder of my computer and the Niryo One:
[image]

Thank you for reading.

mpinol, 2021-05-03:
Hey @TurtleIndigo, 

Thank you for all of the info and the screen shots!

I recently merged a new [doc](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/ros_unity_integration/network.md ) into our dev branch that describes the ROS-Unity network connections that may have some info that might help. 

We should start by figuring out why [Unity cannot communicate with ROS](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/ros_unity_integration/network.md#1-can-unity-send-messages-to-ros). 

Which is the second error you described, 
```
NullReferenceException: Objects reference not set to an instance of an object
(RealSimPickAndPlace.PublishJoints() (at Assets/Scripts_Part4/RealSimPickAndPlace.cs:108)
```

When you run the `roslaunch niryo_moveit part_4.sh` command on the Niryo One, can you confirm whether you see a line output to the console that says something like `Starting server on 10.10.10.10:10000`?

TurtleIndigo, 2021-05-04:
Hey @mpinol,

Thank you for sending me your doc.

I managed to `ping ROS_IP` on my PC computer (with Unity and without ROS):
[image]

This is a screenshot of my Unity console (same PC) with errors quoted in my previous message. I don't have `[UnityTcpSender]: ROS-Unity Handshake received, will connect to UNITY_IP_ADDRESS:5005` message in it:
[image]

In order to run a ROS command, I moved on to my Linux PC (with ROS and without Unity). I also managed to `ping ROS_IP`:
[image]

When running `roslaunch niryo_moveit part_4.sh`, I get `Starting server on 10.10.10.10:10000` but error appears:
[image]

TurtleIndigo, 2021-05-06:
Hey @mpinol,

I managed to fix issues in my Unity console by building and sourcing with a SSH connection. I am new to ROS and did not think about it directly.

I then get `Starting server on 10.10.10.10:10000` when running `roslaunch niryo_moveit part_4.sh` without error:
[image]

I also get `ROS-Unity server listening on UNITY_IP_ADDRESS:5005` in my Unity console:
[image]

When I try to Publish, I get these erros in my Putty terminal, where I launched the SSH connection. It seems that I have a clock issue:
[image]

Do you know how it can be fixed ?

mpinol, 2021-05-06:
Hey @TurtleIndigo,

Great job getting this far!

I don't have access to a Niryo One at the moment so I am going to be operating from my memory.

One thing that stands out to me is that `roscore` is being started when you run the launch file. I believe that this should be started when the Niryo One is turned on so it is a little odd that you are seeing it in the console. Can you confirm whether roscore is running on the Niryo One, before you launch the `part4.launch` file, by running any ROS command? Ex. `rostopic list`

On the Niryo One can you run the following command to make sure that the `robot_description` rosparam was loaded?  `rosparam get robot_description`  This should return the contents of the Niryo One URDF file.

Have you made any changes to the `niryo_moveit` ROS package?

TurtleIndigo, 2021-05-07:
Hey @mpinol,

Thank you!

Two changes have been made to `niryo_moveit` in ROS package:
- Changed IP address, as requested, in `params.yaml` file
- Made the script `server_endpoint.py` an executable script (with the command `chmod +x /PATH/TO/ROS/src/niryo_moveit/scripts/server_endpoint`), as it was not: It allowed me to build and source on the robot catkin workspace, which did not work because of "missing script".

I can confirm that `rostopic list` worked and I think that `rosparam get robot_description` either. Here is what I got:
[image]

I then retried to launch and, miraculously, the clock error disapeared. I then got less errors, the robot moved (oddly) in the Unity scene and the real one was still static:
[image]

Unity side:
[image]

mpinol, 2021-05-11:
Hey @TurtleIndigo ,

Did the simulated Niryo One complete the pick and place task?

TurtleIndigo, 2021-05-17:
Hey @mpinol,

The real and the simulated Niryo One move simultaneously !

I did not think to let Niryo Studio open and connected in order to play the scene in Unity : which seems to be requiered.
Also, real Niryo One position has to be set (on Niryo Studio), as the start position on Unity, before playing the scene to avoid odd movements. In Niryo Studio - Arm Command, move joint at (0,0,0,0,0,0) rad.
It could be added on the tutorial for newbies like me 👍 

Thank you for your help on this issue.

closed 2021-05-17 by TurtleIndigo