Unity-Technologies/Unity-Robotics-Hub #209: Tag version Updates when Installing Unity Robotic Packages

A-Ivan, 2021-04-21:
Hi Unity Robotics Team.

I believe the ROS-TCP-Connector and URDF-Importer tag versions needs to be updated from **v0.2.0** to **v0.3.0** in your quick setup guide to reflect the new version update - https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/quick_setup.md

"
Enter the git URL for the desired package.

For the ROS-TCP-Connector, enter https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#**v0.2.0**.
For the URDF-Importer, enter https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#**v0.2.0**.
"

WalterRainey, 2021-04-23:
I need a trustworthy person 2 give me my government program back.

mrpropellers, 2021-04-26:
Hey @A-Ivan thanks for bringing this to our attention. You should actually be able to simply omit the tag to get the latest version. I'll submit a little fix to clarify in the doc.

closed 2021-04-26 by mrpropellers