Unity-Technologies/Unity-Robotics-Hub #212: [BUG-REPORT] Communication problems when using multiple Unity Editor instances

ja99, 2021-04-30:
**Description:**
When using 2 Unity Instances (one sending, one receiving) the receiving one doesn´t get any messages, if the sending one was started first.

**Steps to reproduce:**
-Make 2 Unity projects, one that sends random points in a self-made point message format, and one that can receive them.
-Give the sending one port 5005
-Give the receiving one port 5006
-Connect both to ROS Noetic on an Ubuntu VM running on the same Windows machine as the 2 Unity instances
-Make sure that both Unity Instances can successfully communicate with ROS
-Make a loopback Python script, that just takes the Messages from the sending instance, puts them on a sperate Topic and sends them off
-Make sure the TCP Subscriber for this seperate topic is added in your server
-Start the transmitting Unity instance
-Start the receiving one

ROS should now receive the Points, and send them back over the spererate topic, but they never land in your receiving Unity instance

**Ugly, but functioning workaround:**
Have a talker script that constantly sends random messages over another throw-away topic to your receiving unity instance. Somehow this makes sure that the points can be received by the receiving Unity instance

Thanks in advance and also thank you for working on such a great project in the first place!

mrpropellers, 2021-04-30:
Hey @ja99 this is a really interesting use case and one we haven't tried yet! Since this is currently unsupported, I'd rather file this as a feature request than a bug. Would you mind closing/recreating this ticket as a feature request with your full use case described so that we can document and prioritize it?  

In the meantime, a more stable workaround might be to create two different endpoints, one for each Unity instance, and configure their publishers, subscribers, and ports accordingly.  Again, totally untested on our end, but might work!

ja99, 2021-05-01:
Hi @mrpropellers, thanks for the quick answer!
I reposted it as feature request, see #214

closed 2021-05-01 by ja99