Unity-Technologies/Unity-Robotics-Hub #213: Is there a way to load ros params in Unity Scripts?

Mingrui-Yu, 2021-05-01:
I have upload some parameters in ROS launch file through <rosparam command="load" ...>. I wonder whether it is possible to get these parameters in Unity Scripts.

Thanks for your help.

at669, 2021-05-03:
Hi @Mingrui-Yu, at this time, there is not a direct way to get the rosparams in Unity. You can use the [Parameter Server]() calls, e.g. `rospy.get_param(param_name)` or `rospy.get_param_names()` to retrieve the parameters and/or values.

From there, you can send them to your Unity instance via publishing them to a topic that Unity is subscribed to, or you can create a custom service that sends a request to ROS and sends back the param names and/or values in the service response, similar to the [ROS/Unity Handshake](https://github.com/Unity-Technologies/ROS-TCP-Endpoint/blob/main/srv/UnityHandshake.srv) implementation in the ROS-TCP-Endpoint.

Hope this helps! If you'd like, feel free to close this ticket and open a new Feature Request with your full use case so we can document and prioritize this feature. Thanks!

Mingrui-Yu, 2021-05-04:
@at669 Thanks for your help.

closed 2021-05-04 by Mingrui-Yu

autonomobil, 2021-07-26:
Is there an update on this topic?