Unity-Technologies/Unity-Robotics-Hub #214: Enabling communication between multiple Unity instances over ROS

ja99, 2021-05-01:
_(Repost as feature request from original Bug-Report #212 )_
**Is your feature request related to a problem? Please describe.**
I don´t want to use RVIZ for visualization, but instead make my own visualization tool in Unity for simulation data, coming from a separate Unity instance.

**Describe the solution you'd like**
I want to be able to connect multiple Unity instances connected to the same ROS server endpoint and send messages bidirectionally.

**Describe alternatives you've considered**
I have found a work-around, where I have a talker script that constantly sends random messages over another throw-away topic to the receiving Unity instance. Somehow this makes the whole thing work.

**Additional context**
A visualization of how something like that could look like:
[image]

at669, 2021-05-03:
Hi @ja99, thanks for reposting this and adding the graphic! I'll file a ticket for this with the provided information, and we will follow up with you. In the meantime, please keep us posted on your progress on this or if you try any other workaround methods!

[ticket# AIRO-649]

ja99, 2021-05-04:
Hi @at669, thanks for the quick answer!
I will probably try @mrpropellers idea of using multiple server endpoints in ROS tomorrow, but for now, the above described workaround is actually working great, even for a single endpoint. The only limitation of it is that I have to start the receiving instance before the sending one.

ja99, 2021-05-05:
Update:
I tried using @mrpropellers idea of using multiple server endpoints in ROS, but I couldn't start both at the same time since they try to use the same IP address.

OSError: [Errno 98] Address already in use

ja99, 2021-07-05:
The solution is to run multiple server endpoints in ros like this:

---------------------------------------------------------------------------------------------------------
```
#!/usr/bin/env python

import rospy

from ros_tcp_endpoint import TcpServer, RosPublisher, RosSubscriber, RosService, UnityService
from robotics_demo.msg import PosRot, UnityColor
from robotics_demo.srv import PositionService, ObjectPoseService

def main():
    ros_node_name = rospy.get_param("/TCP_NODE_NAME", 'TCPServer')
    buffer_size = rospy.get_param("/TCP_BUFFER_SIZE", 1024)
    connections = rospy.get_param("/TCP_CONNECTIONS", 10)
    tcp_server = TcpServer(ros_node_name, buffer_size, connections, tcp_port = 10001)
    rospy.init_node(ros_node_name, anonymous=True)
    
    tcp_server.start({
        'pos_rot': RosPublisher('pos_rot', PosRot, queue_size=10),
    })
    
    rospy.spin()

if __name__ == "__main__":
    main()
```

----------------------------------------------------------------------------------------------------

Where you just itereate through the port numbers.

If you also need multiple connections from one Unity instance, you can add multiple Ros Connections to your scene and adjust the ports there as well. When you use them in code then, you simply add them through a serialized field, instead of using the built in singelton-way you see in the tutorials.

You can see an example of this in this project: https://github.com/ja99/LidarCameraSimUnity.git 
where the lidar and camera use different connections in the same unity instance to raise performance.

closed 2021-07-05 by ja99