Unity-Technologies/Unity-Robotics-Hub #215: Unable to build, generated messages could not find suitable override and namespace

ctzy818, 2021-05-04:
Hi,

I followed the ROS–Unity Integration: Publisher tutorial and got the code provided to work on the Unity Editor. I have since tried to write my own msg files and a publisher, using the code provided in the tutorial as a template.

Tutorial link: https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/publisher.md

However, I am not able to build the project from Unity. The project still includes the content from the tutorial, and those codes also produce the same error. I saw the comment in the generated message codes saying that we should not be editing the codes, thus may I know if I have missed any other packages/steps/version mismatches that may have caused this issue? 

**Console errors (with ref to the MUnityColor msg, others are similar):**
Assets\RosMessages\RoboticsDemo\msg\MUnityColor.cs(34,38): error CS0115: 'MUnityColor.SerializationStatements()': no suitable method found to override
Assets\RosMessages\RoboticsDemo\msg\MUnityColor.cs(45,29): error CS0115: 'MUnityColor.Deserialize(byte[], int)': no suitable method found to override
Assets\RosMessages\RoboticsDemo\msg\MUnityColor.cs(59,32): error CS0115: 'MUnityColor.ToString()': no suitable method found to override
Assets\RosMessages\RoboticsDemo\msg\MUnityColor.cs(6,13): error CS0234: The type or namespace name 'Robotics' does not exist in the namespace 'Unity' (are you missing an assembly reference?)
Assets\Scripts\RosPublisherExample.cs(5,13): error CS0234: The type or namespace name 'Robotics' does not exist in the namespace 'Unity' (are you missing an assembly reference?)
Assets\Scripts\RosPublisherExample.cs(13,5): error CS0246: The type or namespace name 'ROSConnection' could not be found (are you missing a using directive or an assembly reference?)
Assets\RosMessages\RoboticsDemo\msg\MUnityColor.cs(10,32): error CS0246: The type or namespace name 'Message' could not be found (are you missing a using directive or an assembly reference?)

**Screenshots**
Console:
[image]

Build settings:
[image]

**Environment Details:**
- Unity Version: Unity 2020.3.4f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic

**Additional context**
After going through the other issues in the GitHub, I came across a comment by another user with a similar issue. Here is the link: https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/53#issuecomment-773277958

Ps. do tell me if more details is required. I am relatively new to Unity and posting issues on GitHub, thus I might have missed out on details/ not know the conventions very well. Thank you!

at669, 2021-05-04:
Hi @ctzy818, thanks for the detailed report and screenshots! Unfortunately, this package is not actively supporting UWP builds, which is causing those error messages during the build process (I'm assuming this still works in the Editor, but please correct me if I'm wrong!). However, you should be able to make a few small changes to the package structure in order to resolve these error messages and create a UWP build.

1) In Unity's Project window, right click on the package `ROS TCP Connector`, and select `Show in Explorer`. This should open an OS folder window called `PackageCache`.
2) Do the same for your project's `Assets` folder. This should open your project's root folder.
3) Within the root folder window, open the `Packages` folder.
4) Move the folder `com.robotics.ros-tcp-connector@<miscellaneous-characters>` from your `PackageCache` window into the `Packages` folder. This will enable you to make and save changes to the package.

Now, in Unity, you should be able to go into the `Packages/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef` file and tick the "Universal Windows Platform" checkbox so that it will be included in the desired builds.

Please view #169 for the source of this solution, and let me know if this resolves your issue.

ctzy818, 2021-05-05:
Dear Amanda,

Thank you for the prompt response. I'll give it a try and update you again.

Yours sincerely,
Chris

ctzy818, 2021-05-06:
> Hi @ctzy818, thanks for the detailed report and screenshots! Unfortunately, this package is not actively supporting UWP builds, which is causing those error messages during the build process (I'm [...]

Hi @at669, I have tried this and it has resolved my issue. Thank you so much!!

at669, 2021-05-06:
@ctzy818, glad to hear it helped! I will close this ticket since the issue is resolved, but feel free to open another if a new issue crops up!

closed 2021-05-06 by at669

ctzy818, 2021-05-21:
Hi Amanda,

Could I ask if the Unity Robotics Hub works in Unity 2019?

Yours sincerely,
Chris

at669, 2021-05-21:
Hi @ctzy818, all of our Unity Robotics packages (ROS-TCP-Connector, URDF-Importer) are expected to work, at minimum, on Unity Editor versions 2020.2 and higher.