Unity-Technologies/Unity-Robotics-Hub #216: Running into issues with custom URDFs

NathanielRose, 2021-05-06:
**Describe the bug**
I have created a URDF file in which I use the UR10 with the Robotiq gripper for pick and place. The UR10 alone works fine but once I add the robotiq gripper I receive these .NET failures.

**To Reproduce**

https://user-images.githubusercontent.com/8294326/117325761-ed99d880-ae88-11eb-8bb8-05d8388a71c6.mp4

**Console logs / stack traces**
```
UnassignedReferenceException: The variable ur10 of UR10TrajectoryPlanner has not been assigned.
You probably need to assign the ur10 variable of the UR10TrajectoryPlanner script in the inspector.
UR10TrajectoryPlanner.Awake () (at Assets/Scripts/UR10TrajectoryPlanner.cs:212)

Invalid worldAABB. Object is too large or too far away from the origin.

Invalid AABB a

Assertion failed on expression: 'IsFinite(d)'
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

Assertion failed on expression: 'IsFinite(outDistanceForSort)'
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

Assertion failed on expression: 'IsFinite(outDistanceAlongView)'
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

Invalid AABB aabb
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

Invalid AABB aabb
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

Invalid AABB aabb

No data available on network stream after 10 attempts.
UnityEngine.Debug:LogError (object)
ROSConnection/<SendServiceMessage>d__21`1<RosMessageTypes.RosTcpEndpoint.UnityHandshakeResponse>:MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@ab45157ad3/Runtime/TcpConnector/ROSConnection.cs:129)
UnityEngine.UnitySynchronizationContext:ExecuteTasks ()

Invalid AABB a
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

Invalid AABB a
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

```

**Expected behavior**
Move UR10 with Gripper with keyboard freely without being dismounted.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.2.0f1]
- Unity machine OS + version: [e.g. Windows 10]
- ROS machine OS + version: [e.g. Ubuntu 18.04, ROS Noetic]
- ROS–Unity communication: [e.g. Docker]
- Branch or version: [e.g. v0.2.0]

**Additional context**
Add any other context about the problem here.

at669, 2021-05-07:
Hi--I should have been more clear. Could you send the full URDF package, including the meshes?

And just to clarify, `Branch or version: Latest` refers to the URDF-Importer's `v0.3.0`?

Finally, did you have any luck investigating internal collisions?

NathanielRose, 2021-05-12:
@at669 Thanks for the quick reply.
Here is the URDF I am using for the UR10 + Gripper https://github.com/NathanielRose/UniversalRobot10_RobotiqGripper/blob/main/ur3_with_gripper/ur10_with_gripper.urdf

Additionally, I am currently using
`https://github.com/Unity-Technologies/URDF-Importer.git#v0.1.2` Is this deprecated?

at669, 2021-05-12:
Thanks! We'll investigate the package linked and update you on that. 

In the meantime, yes, `v0.1.2` is out of date, and `v0.3.0` is the currently most stable version. Please try updating the package version and let us know if the behavior is different.

NathanielRose, 2021-05-12:
Just updated, still receiving that behaviour with v0.3.0

https://user-images.githubusercontent.com/8294326/118032143-3822c580-b35f-11eb-8201-8d768746da76.mp4

NathanielRose, 2021-05-22:
@at669 Any follow up on the UR10 Gripper URDF and why it could be failing?

at669, 2021-05-24:
Hi, the team has not yet been able to investigate this issue yet, but the ticket has been filed internally and will be investigated as soon as possible. In the meantime, I would suggest investigating the potential [internal collisions](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md#disable-collision-support) and disabling them to see if that can resolve your issue. Thanks for your patience!

at669, 2021-06-21:
Hey @NathanielRose, sorry for the late response! I'm unable to recreate your issue using your provided URDF package and with the following setup:

```
Unity version: 2020.2.0f1
URDF-Importer version: v0.4.0
Physics solver: TGS
URDF-Importer mesh decomposer: VHACD
UR10 Controller configuration:
    Stiffness: 10000
    Damping: 1000
    Force limit: 1000
    Speed: 30
```
The physics solver, if not already TGS, can be set under `Edit > Project Settings > Physics > Solver Type > Temporal Gauss Seidel`--all other Physics settings were left as default. Additionally, the URDF-Importer has recently been updated to `v0.4.0`, which I would suggest updating to! Finally, TGS exposes some internal collisions in the gripper. You may want to add the following lines to your custom URDF:

```xml
<disable_collision link1="left_inner_knuckle" link2="left_outer_knuckle"></disable_collision>
<disable_collision link1="right_inner_knuckle" link2="right_outer_knuckle"></disable_collision>
<disable_collision link1="left_inner_knuckle" link2="left_inner_finger"></disable_collision>
<disable_collision link1="right_inner_knuckle" link2="right_inner_finger"></disable_collision>
<disable_collision link1="left_inner_knuckle" link2="robotiq_arg2f_base_link"></disable_collision>
<disable_collision link1="right_inner_knuckle" link2="robotiq_arg2f_base_link"></disable_collision>
```

https://user-images.githubusercontent.com/31416491/122823659-df6b1480-d29c-11eb-9a27-90f16a9f0884.mov

Let me know if this set of setup configs resolves your issue.

NathanielRose, 2021-06-28:
@at669 Than you! The collision disabling I believe fixed the issue with movement in space. I am following the [code](https://www.youtube.com/watch?v=D3Ya_RgazlE) for opening and closing the gripper and not seeing the correct behaviour for closing a robotiq gripper. This can be a separate issue if this may be a completely tangential thing. I was curious if the way we configured the collision disabling is causing the gripper to not close correctly?

https://user-images.githubusercontent.com/8294326/123613568-73882f00-d7fb-11eb-8aa8-efae5d9309f5.mp4

at669, 2021-06-28:
The collision disabling tags are the same ones as used in the [Object Pose Estimation](https://github.com/Unity-Technologies/Robotics-Object-Pose-Estimation) project, which is currently configured similarly (i.e. TGS, VHACD mesh decomposition, same collision disabling tags). 

I think you might have added the wrong link for the code!

NathanielRose, 2021-06-28:
@at669 So embarassing! Apologies for that! [Code](https://github.com/Unity-Technologies/Robotics-Object-Pose-Estimation/blob/main/PoseEstimationDemoProject/Assets/TutorialAssets/Scripts/TrajectoryPlanner.cs#L72-L83) is linked below:

```
    public IEnumerator IterateToGrip(bool toClose)
    {
        var grippingAngle = toClose ? gripperAngle : 0f;
        for (int i = 0; i < gripperJoints.Count; i++)
        {
            var curXDrive = gripperJoints[i].xDrive;
            curXDrive.target = multipliers[i] * grippingAngle;
            gripperJoints[i].xDrive = curXDrive;
        }
        yield return new WaitForSeconds(jointAssignmentWait);
    }
    ```
    
    I played with the `grippingAngle` to see if that was the root cause for the left finger remaining dormant during gripping and it didn't change much. This is a minor issue and I can take it offline to investigate but thought I would raise it on this thread if you have seen this behavior before. Thanks again!

at669, 2021-06-28:
I'd expect the gripping angle to modify the movement significantly, so that seems a bit strange! Aside from that value, I'd suggest verifying the stiffness, damping, and force limits on the gripper's articulation bodies as well, and changing their target values in the inspector to see if they move as expected. Feel free to close this ticket and open a new one if the original issue appears to be solved and open a new one if you can't seem to get the gripper working as expected though!

hyounesy, 2021-07-08:
Hi @NathanielRose. Just wanted to verify that your issue has been resolved before closing the issue.

NathanielRose, 2021-07-08:
It has been resolved, Thanks again for the support here!

closed 2021-07-08 by NathanielRose